MarlinFirmware/Marlin #27644: [FR] Add additional extruder retraction feedrate limit

FlyingRatBull, 2025-01-15:
### Describe the feature you want

My printer's extruder has a different maximum speed it can handle when extruding vs. retracting stemming from the fact that it is harder to push the filament through the nozzle than to pull it back to the spool.

Currently I can set the extruder feedrate limit through `M203 E25` or the LCD menu. However with this approach I also limit the speed of the retraction which in my case increases blobs through to slow retraction and also increases the print time significantly (my max extrusion speed is only *5mm/s* vs retraction *>25mm/s* thus increasing the retraction time from *0,04s* to *1s* for **each retraction**).

Alternatively I could limit the printing speed but this would also limit the printer unnecessarily as a *0.4mm* line would print at most as fast as a *0.8mm* line however as the limiting factor is the extruder, the 0.4mm line could be printed twice as fast when just adhering to the extruder limits.

I suggest to add a separate speed limit for extruder retraction.

Possible ways to implement:

* Add a config value like `EXTRUDER_RETRACTION_LIMIT_MULTIPLIER` to calculate the retraction limit by multiplying the feed rate with this factor.
  * Relatively easy to implement
  * Would not be that flexible as for example different feed rate limits for different nozzles would then still impact the retraction limit (when not changing the config for every nozzle change) - although not as much as currently anymore.
* Extend `M203` to accept an R value for extruder retraction.
  * Harder to implement
  * Would provide the most flexibility and thus the best possible speed of the printer as one could just call `M203 E5 R25` on a nozzle change and get the most out of their printer.

### Additional context

Why I would like this feature in Marlin and not in a slicer:
  * This is directly tied to the hardware limits of the printer and thus makes sense to have in firmware
  * Code generators (like for calibration) can - in general - not differentiate between extrusion speed and retraction speed and you want calibrations such as linear advance to be as close to the actually used printer profile as possible

I can't just run my extruder on the retraction speed as it doesn't just go slower/as fast as it can, it actually grinds the extrusion gear and no extrusion happens at all.

thisiskeithb, 2025-01-16:
Duplicate of #27565

closed 2025-01-16 by thisiskeithb

FlyingRatBull, 2025-01-16:
@thisiskeithb Isn't #27565 about e-**jerk**, so the change in acceleration (mm/s³)? If so, that wouldn't help me as neither the acceleration nor the change in acceleration would change anything if the retraction still caps out at the extrusion **speed** (5mm/s) instead of 25mm/s.