MarlinFirmware/Marlin #27647: unit test / unity.h compilation errors

mtraven, 2025-01-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When attempting to compile the most recent bug fix (as of yesterday) and the beta release I am unable to complete compilation.  I hesitate to call this a bug, given that I cant compile either of these versions "out of the box", or for my environment (btt skr 2).  I must have something wrong?   I've attached pictures of the errors for these versions.  If I go back to 2.1.2.5 (latest stable), everything works fine.  I also dont find the "test" folder that seems to be giving me problems in the newer versions.  I'm not sure if thats something corresponding to the newer versions, or just on the bugfix / beta branches?

One thing I noticed that might help: I had to change my environment from BIGTREE_SKR_2_F429_USB to STM32F429VG_btt.  Not a big deal, seems like there was just some nomenclature change...but maybe those changes are related?

[image]

[image]

[MarlinConfig.zip](https://github.com/user-attachments/files/18462684/MarlinConfig.zip)

### Bug Timeline

new staring with current bf & beta versions

### Expected behavior

I expected it to compile!

### Actual behavior

it did not compile!

### Steps to Reproduce

1. unzip current marlin repo.
2. modiy platformIO.ini for STM32F429VG_btt environment
3. change motherboard to BOARD_BTT_SKR_V2_0_REV_B
4. build.

### Version of Marlin Firmware

bugfix 2.1.x (16 jan 2025)

### Printer model

homemade

### Electronics

btt skr-2 with 2209 drivers

### LCD/Controller

btt mini 12864

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

IdeaMaker

### Host Software

Repetier Host

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

the config files I've included are very basic, just enough to get it to compile for my system.

thanks  to all involved in maintaining this project, it cant be easy!

ellensp, 2025-01-18:
Marlin uses source filtering, In features.ini it has 
MARLIN_TEST_BUILD                      = build_src_filter=+<src/tests>

Ie the tests folder is only compiled if you have enabled MARLIN_TEST_BUILD

My best guess if you have totally broken the platformio.ini file disabling source filtering, you probably used import arduino or some other loading option instead of just opening the Marlin folder with menu File|Open folder. 

BIGTREE_SKR_2_F429_USB is not a valid build environment for this board, it was renamed to STM32F429VG_btt_USB 6 months ago
This is the list of valid build environments for this motherboard

<ul>
<li>STM32F407VG_btt</li> 
<li>STM32F407VG_btt_USB</li> 
<li>STM32F407VG_btt_USB_debug</li> 
<li>STM32F429VG_btt</li>
<li>STM32F429VG_btt_USB</li> 
<li>STM32F429VG_btt_USB_debug</li>
</ul>

If your using the stock "Platformio: Build" button you have to set default_envs to one of these build environments in platformio.ini 
Or you use the Auto build Marlin plugin which will give menu options on what you can use

[image]

restore your platformio.ini file to the stock one 
https://raw.githubusercontent.com/MarlinFirmware/Marlin/refs/heads/bugfix-2.1.x/platformio.ini

closed 2025-01-18 by ellensp

mtraven, 2025-01-18:
why did you close my topic before I even had a chance to respond?  Seems rather dismissive.

I seem to have given you the impression that I dont understand how to setup a platformIO environment.  I do, I only pointed out the nomenclature change in case it was related to another change.  

I've not done any of the things you've accused me of:
-ive only changed the most basic of configuration items in an attempt to get a successful compilation (env, board, serial, screen for feedback)
-I have started over with the entire repo several times, so the platformIO.ini is not anymore broken than it is out of the box and its been restored to the original a half dozen times.
-i even download a fresh repo of the BF branch today before I responded, same results.
-I've not imported an arduino project, or any of that.  Just opening the repo I get from github.  I do save the vscode workspace for easy opening in the future, but thats never been a problem in the past.  I dont use marlin auto build, never really saw the point.

I can confirm that default`_src_filter = +<src/*> -<src/config> -<src/tests>` is in the platform.ini and that `MARLIN_TEST_BUILD = build_src_filter=+<src/tests>` is in features.h

all that being said, I did find the problem.  My printer firmwares all live(d) in a folder called "[3dprinter]" ..that hadn't been an issue for any previous version of marlin, but now seems to be.  Removing the "[ ]" from the folder name resolved the issue.

now its closed.

thinkyhead, 2025-03-27:
Sorry that the early closing appears dismissive. Every volunteer on the project has a different style. When an issue is understood by a volunteer to be externally caused, it may be more convenient for some busy lives to close the issue right away, lest it be forgetten. It's not meant to be dismissive. It's just coldly administrative.

I generally recommend that our volunteers aim to avoid these kinds of appearances and be sensitive to users at every level. So, I will reiterate that general advice here as well. Please be sensitive to users at every level. According to Adler the personal touch is best, and I tend to agree.