MarlinFirmware/Marlin #27651: SENSORLESS_PROBING FOR COREXY BEDS

sinus19, 2025-01-23:
### Is your feature request related to a problem? Please describe.

No

### Are you looking for hardware support?

No

### Describe the feature you want

It would be a great to take advantage of this feature for those of us that use 60-100c enclosure temps where traditional ABL sensors do not work well. I understand that it can put a lot of stress on the lead screws (for those who use screws). However, I am using Quad 16mm Ball Screws and they are plenty rigid and strong enough to take on the stress. The gantry is also very rigid and heavy.

I noticed this feature is primarily used in Delta Printers and utilizes the X,Y, & Z Diag pins. I would assume you guys would have to add additional modules? (excuse my poor lack of coding terminology) where it can be setup to utilize all of the Z Diag/Endstops defiend in the pins directory. In my case with 4 independent Zs on stallgaurd, I redefined the Diag and motors as Z, Z2, Z3 & Z4 in the pins directory for my Octopus Pro V1.1.  

Lastly, It would also be nice to have the ability to adjust the StallGuard Sensitivity for probing. Especially for those that may not have such a robust gantry or Z axis. You can tune the value to best fit your situation 

### Additional context

This feature would be a game changer if we can get it to work on Beds that travel in the Z! 1 less cable and 1 less piece of hardware. Build plate surface would not matter, Z offset would be simpler to adjust or in some cases could be set to zero. My printer is a custom 7 axis, 16mm ball screw with all 4 Zs fully independent and on StallGaurd. The Z homes accurately every single time and has been extremely reliable. I would be happy to be the testing mule if needed!

rondlh, 2025-01-23:
For Marlin it doesn't matter if the bed or the gantry moves, that should not be the issue here.

What you want might already be supported (SENSORLESS_PROBING), but in your setup I doubt it will have the results you are hoping for. Just imagine what happens when you probe on different locations on the bed, this will affect the 4 Z-steppers in a different way. 
One time Z1 might be triggered first, the next time Z2 is first, this would impact the accuracy because the bed would tilt a bit. This is even neglecting that sensorless probing is not very accurate and repeatable in the first place. I doubt you can reach anywhere near the accuracy required for a good first layer.
I guess a load-cell in the head or under the bed could work in your setup.

sinus19, 2025-01-23:
If you look cloesly into the post processing pin for that feature, there is a line that says something along the lines of "looks at all Diag pins while Probing..." this tells me that theres already some sort of algorithm that chooses one of the 3 Diag pins (Delta printer) to register the hit/probe depending on the location of the nozzle.

Here is a video of a guy who uses it on a Delta. So we know it works. And i think the Developers can polish it up 

https://youtu.be/xb8dM9duGOs?si=2-uwrqO1NdDfe9zf

thisiskeithb, 2025-01-31:
`SENSORLESS_PROBING` is not restricted to Deltas.

If something is not working, please create a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) and be sure to attach your configs.

closed 2025-01-31 by thisiskeithb

sinus19, 2025-01-31:
> > https://youtu.be/xb8dM9duGOs?si=2-uwrqO1NdDfe9zf
> 
> Nice video, it seems to work quite well, I wonder how repeatable it is. The bed is fixed, which means it cannot tilt. In your case the bed might tilt depending on where you hit the bed, one side will go down and the other will go up.
> 
> Anyway, I just enabled SENSORLESS_PROBING and defined the sensitivity for the Z-steppers. I have 4 Z-motors (moving bed) but the motion system is XY, not CoreXY. No errors are reported when I compile like this, which makes me believe you could give it a try. What is actually stopping you?
> 
> If you do, make sure to put some protection on the bed and a cover on the nozzle so you can find the right sensitivity values. Start testing from the very sensitive side of the values, which cause an immediate (false) trigger. And use a low homing current for all Z-motors (Z_CURRENT_HOME). Depending on which TMC stepper you use the meaning of the sensitivity values are different: LESS SENSITIVE TMC2209 <-- [ 0...255] --> MORE SENSITIVE MORE SENSITIVE TMC5160 <-- [-64....63] --> LESS SENSITIVE
> 
> I also found the sensitivity depends on the room temperature, so be careful.

I too have a Zs, 16mm ball screws. bed is rock solid. i couldn't get it to register a probe. regardless of the sensitivity.Im on 2.1.2.4 what version are you on?