MarlinFirmware/Marlin #27660: [BUG] Errors compiling marlin for MKS Tinybee

theyeetmaster992, 2025-01-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to compile marlin for mks tinybee but get continuous errors with using provided firmware and new firmware. Trying to find a version that compiles successfully but with not much success so far. Seems to also have compilation issues when MKS mini 12864 v3 is selected. 

https://github.com/makerbase-mks/MKS-TinyBee/issues/125

### Expected behavior

Expected to get a successful compile when instructions where followed or changeable minor issues to be defined or undefined.

### Actual behavior

Compilation failure with many warnings and partitions.bin created instead of firmware.bin and .elf

### Steps to Reproduce

Attempt compile firmware provided by mks for mks tinybee

### Version of Marlin Firmware

any

### Printer model

any

### Electronics

upgrade board

### LCD/Controller

MKS 12864 mini v3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Desktop.zip](https://github.com/user-attachments/files/18617674/Desktop.zip)

ellensp, 2025-01-31:
You need to use latest bugfix 2.1.x 

That libraries where update with raspberry pi 2040 code and  broke thins

Only bugfix has this patch
https://github.com/MarlinFirmware/Marlin/commit/29635232d356175fee4a3383cafa7a967f786286

theyeetmaster992, 2025-01-31:
latest bugfix was tried and doesn't work

ellensp, 2025-01-31:
attach butfix config files

ellensp, 2025-01-31:
first issue I see in your config files

#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

but tinybee doesnt have a -1 port.  

the usb cable is connected to CH340C chip then to uart 0 on the esp32
so #define SERIAL_PORT 0 is correct but #define SERIAL_PORT_2 -1 is not and should be removed

theyeetmaster992, 2025-01-31:
its for the built in wifi i believe

theyeetmaster992, 2025-01-31:
> No that use a socket (internal to chp communication)

so built in esp32 wifi doesn't need a serial assigned for it at all?

ellensp, 2025-01-31:
#define MKS_MINI_12864_V3
#define NEOPIXEL_LED
#define NEOPIXEL_TYPE          NEO_RGB
#define LED_CONTROL_MENU
#define LED_USER_PRESET_STARTUP
option but I like it to confirm rgb is working
#define NEOPIXEL_STARTUP_TEST

theyeetmaster992, 2025-01-31:
Hmm ok, do i just add those somewhere in conf.h?

theyeetmaster992, 2025-01-31:
[image]

might be getting closer

theyeetmaster992, 2025-01-31:
> #define MKS_MINI_12864_V3 #define NEOPIXEL_LED #define NEOPIXEL_TYPE NEO_RGB #define LED_CONTROL_MENU #define LED_USER_PRESET_STARTUP option but I like it to confirm rgb is working #define NEOPIXEL_STARTUP_TEST

#define LED_USER_PRESET_STARTUP seems to be for a secondary neopixel and requires NEOPIXEL2_SEPERATE to be active

theyeetmaster992, 2025-01-31:
Didn't receive a success when compiling completed but will assume it was. Think putting marlin in root directory is a must now. Will update if flashing has been successful and machine working.

theyeetmaster992, 2025-01-31:
[image]

Enabled wifi in conf_adv.h

ellensp, 2025-02-01:
LED_USER_PRESET_STARTUP only needs NEOPIXEL2_SEPERATE if you enabled NEO2_COLOR_PRESETS

theyeetmaster992, 2025-02-01:
[Marlin (2).zip](https://github.com/user-attachments/files/18627611/Marlin.2.zip)

Wifi doesn't seem to work. Is there something else that needs to be define to make it be picked up?

[image]
Also how to get rid of this.

theyeetmaster992, 2025-02-01:
how does configuration_secure.h work? must you enter the network its going to connect to or what network it is going to be itself?

ellensp, 2025-02-01:
connect to your wifi

The code i Marlin`
``
void wifi_init() {

  SERIAL_ECHO_MSG("Starting WiFi...");

  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PWD);

```

these are the wifi modes, 
WiFi.mode(WIFI_STA) | station mode: the ESP32 connects to an access point
WiFi.mode(WIFI_AP) | access point mode: stations can connect to the ESP32
WiFi.mode(WIFI_AP_STA) | access point and a station connected to another access point

theyeetmaster992, 2025-02-01:
[image]

I mean by this if i where to try and use it

theyeetmaster992, 2025-02-04:
I have gotten the machine this was for to move without anything burning, have yet to check the wifi. Screen and leds work and goes through startup led check

ellensp, 2025-02-05:
you do require #define SERIAL_PORT_2 -1 

by default the ESP3D for Marlin ESP32 interface is a access point. 
Default SSID: MARLIN_ESP
Default password: 12345678
Default IP: 192.168.0.1

I have cleaned up all the warning I can In PR https://github.com/MarlinFirmware/Marlin/pull/27672

Cannot do anything about the last two 
[image]

As that is part of the framework provided by expressive, and outside of Marlin Control

closed 2025-02-05 by ellensp

theyeetmaster992, 2025-02-05:
Thank you very much for looking into it. I ended up using the serial settings specified for it as i have found a esp32 GitHub resource that explained it better in its description with additional marlin settings, have yet to test it fully.