MarlinFirmware/Marlin #27661: [BUG] Marlin 2.1.3 Beta2 fails to build for FTDI EVE display

maralb1970, 2025-02-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am one of the few running a printer with a FTDI EVE display attached. Marlin 2.1.3 Beta 2 fails to build for 3D printer with an FTDI EVE based screen (Fystec TFT81050).

It throws an error 'LEVELING_COMMANDS' not declared - did you mean 'BED_LEVELING_COMMANDS'? for /extui/ftdi_eve_touch_ui/generic/leling_menu.cpp

Changing LEVELING_COMMANDS into BED_LEVELING_COMMANDS as suggested by  the complier makes the firmware to build successfully.

[image]

### Bug Timeline

It was fine in Marlin 2.1.2.5 and before.

### Expected behavior

I expect the firmware to build successfully.

### Actual behavior

Firmware failed to build

### Version of Marlin Firmware

Marlin 2.1.3 Beta 2

### Printer model

Hiprecy LEO

### Electronics

Fystec S6 V2.0

### LCD/Controller

Fysetc FTDI EVE TFT81050

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/18628027/Marlin.zip)

ellensp, 2025-02-01:
was updated in https://github.com/MarlinFirmware/Marlin/pull/27275

from

`case 2: SpinnerDialogBox::enqueueAndWait(F("G34")); break;
`
to

`case 2: SpinnerDialogBox::enqueueAndWait(F(LEVELING_COMMANDS)); break;
`

Yes it looks like LEVELING_COMMANDS was not defined.

ellensp, 2025-02-01:
Issue is lulzbot has that define as an additional define to stock in their Configuration_adv.h 

Eg
https://gitlab.com/lulzbot3d/marlin/-/blob/master/Marlin/Configuration_adv.h#L4149

ellensp, 2025-02-04:
pr created, closing

closed 2025-02-04 by ellensp