MarlinFirmware/Marlin #27675: [BUG] Optical endstops not working on MKS Tinybee

theyeetmaster992, 2025-02-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Optical endstops not working/being picked up by marlin on MKS Tinybee. Tried swapping cables and individual wires, still not working. Could be config issue or issue on another file like pins(not sure honestly)

### Expected behavior

Expected optical endstops to work like on the old motherboard. Will show red light when working.

### Actual behavior

end stops don't trigger and machine grinds

### Version of Marlin Firmware

2.1.x bugfix

### Electronics

board upgrade

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin (3).zip](https://github.com/user-attachments/files/18677156/Marlin.3.zip)

ellensp, 2025-02-06:
mechanical and optical endstops both work the same way as far as the controller is concerned.

The endstop sends a high or a low depending on is current state.

You just tell marlin if it is triggered high or low.

Any other differences are electrical. 

measure the voltage on the signal pin, does sit change when you sick a piece of cardboard the slot?

Also note that most 3d printed plastics are IR transparent, ie you cannot interrupt an optical endstop with them.

theyeetmaster992, 2025-02-06:
I will check it when i have more time. Parts are injection molded abs plastic/solid.

closed 2025-06-23 by github-actions[bot]