MarlinFirmware/Marlin #27676: [BUG] (bug summary)Using Z_MAX for Z_MIN_PROBE disables Z_MIN_ENDSTOP Operation in MARLIN 2.0.x

rpweaverjr, 2025-02-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When not using Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN (I'm using Z_MAX, P1_29) and not using SERVER 0 in the BLTouch configuration (I'm using SERVO 3, P4_28) along with associated hardware connections, when leveling (B29) the Z_MIN_ENDSTOP_PIN seems to be disabled and the printhead attempts to drill through the buildplate. The 3DTouch gets triggered (M119, Z_MAX and  Z_MIN_PROBE) but there is a probing error reported in Pronsole. The Z_MIN_ENDSTOP_PIN (Ender 3 switch) should stop the Z movement even if the probe fails but it doesn't. Outside of leveling, everything looks fine. M401 deploys the probe and M402 stows the probe. M119 shows open when it is deployed and TRIGGERED when it is stowed. When homing (B28), Z MIN ENDSTOP switch stops the Z movement.
I am not yet able to build 2.1.x. MKS doesn't or has not yet supported this board beyond 2.0.x. MKS SGEN V1.0. I get a no Motherboard when I try to autobuild using the configuration that works with 2.0.x.
Running a g28 (first time always works) followed by a m401 followed by a g28 will attempt to drill through the buildplate.

### Bug Timeline

I've been running Bugfix 2.0.x for years without the 3DTouch. I the past weeks I've been trying to use the 3DTouch with the configuration that worked as a baseline.

### Expected behavior

First, I expect the Z_MIN_ENDSTOP switch to always stop the -Z movement even if the probe fails when leveling like it does when homing which doesn't always stop either.
I expect the probe to stop the Z movement when it is triggered.

### Actual behavior

Z movement doesn't stop when leveling until  the printbed stops and the stepper grinds and stalls. It stops fine when homing.
The probe triggers when leveling but Pronsole reports a probing error. It doesn't appear to be read.
Running a g28 followed by a M401, followed by a G28 will also attempt to drill through the build plate. Both Z_MIN_PROBE and Z_MIN are triggered.
SENDING:M119
Reporting endstop status
x_min: open    [37m 18°>←[00m
y_min: open    [37m 18°>←[00m
z_min: TRIGGERED37m 18°>←[00m
z_max: TRIGGERED37m 18°>←[00m
z_probe: TRIGGEREDm 18°>←[00m
←[01m←[32mcom7←[37m 18°>←[00m
Had to power down to stop. This is after repowering.

### Steps to Reproduce

1. Using Z_MAX as the probe pin and Servo 3 as the BLTouch connections.
2. run G28 and everything works normally. The printhead moves to the homing position and everything stops.
3. run G29 and the printhead moves up and over to the first leveling location. then it lowers until the nozzle touches the bed and it starts to grind. The probe is triggered and the Z switch is triggered.
4. Second error but probably related, Run a G28
5. m401
6. g28

### Version of Marlin Firmware

SENDING:M115 FIRMWARE_NAME:Marlin bugfix-2.0.x (GitHub) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Creality Ender-3 Pro 

### Electronics

MKS SGEN V1.0 3DTouch from Amazon

### LCD/Controller

MKS TFT 32, Creality 12864 LCD doesn't seem to work with this board.

### Other add-ons

3DTouch from Amazon

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configurations.zip](https://github.com/user-attachments/files/18682993/Configurations.zip)

thisiskeithb, 2025-02-06:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

rpweaverjr, 2025-02-07:
If you deploy the probe and make a bunch of small -Z moves until the probe is triggered, it will not respond to more -Z moves. It will respond to +Z moves. If instead, you make a large -Z move and the probe is triggered, it will not stop until the long Z move is complete regardless of the current Z position (it will attempt to go through the print bed). 
G28 will release the block on the -Z moves.

This tells me that if the -Z move during leveling are multiple small moves, the probe should be triggered and subsequent -Z moves will no longer cause the printhead to crash into the print bed.

rpweaverjr, 2025-02-09:
Unfortunately, so far,  I can't get anything other than the one I have to build. I've been trying for weeks. I thought I bricked the board with some of these builds but luckily, the old firmware brought it back to life.  I am a retired computer engineer but I am not familiar with the coding conventions used in Marlin. It takes me a long time to fix build errors when I can. Most recently, I've done extensive debugging in Visual Basic and Visual Studio (not VScode). I am also familiar with C and some C++. If I could get someone to help me with getting it to build, I would be grateful. Unfortunately, MKS doesn't seem to be interested in this old board since they are no longer available. When I use the available MKS SGEN related code, it is clear, this was never built let alone tested. It may be possible to build the SGEN-L (based on the LPC1768 instead of LPC 1769) but I haven't tried. Supposedly, they are very similar.
PIO is also a mystery and it will use non matching projects/versions without complaining. I think that is where I got into trouble.
In addition to what I have working, I've tried to build 2.0.5.3, 2.0.9.7, 2.1.2.1 and Bugfix 2.1.x without success.