MarlinFirmware/Marlin #27679: [BUG] BABYSTEPPING it does not work

ppmanzionna, 2025-02-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,
I tried marlin beta (2.1.3b1, 2.1.3b2) and bugfix 2.1.x in a Artllery Hornet with examples configuration, but babystepping does not work. With Marlin 2.0.9.7, no problem.

### Expected behavior

Babystepping it does not work.

### Version of Marlin Firmware

2.1.3b2, 2.1.3b1, bugfix 2.1.x

### Printer model

Artillery Hornet

### Electronics

BOARD_ARTILLERY_RUBY

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thinkyhead, 2025-02-19:
@ppmanzionna

Babystepping is working for me so I'm not sure what's happening there. Is there no babystepping at all on X, Y, or Z (before and after homing)? See if you get any movement using a smaller babystep on Z, such as 0.05 or 0.02. You should also enable `EP_BABYSTEPPING` and test the commands `M293`/`M294`. If none of this results in any babystepping then we can look into the history to locate the change that broke babystepping for your type of setup.

thinkyhead, 2025-02-19:
> How can I check the history to find the change that stopped babystepping?

The basic strategy is to start by finding a commit from some point in the "distant" past where the feature works. So you should try 2.1.2.5 to see if it works there. If not, you can go back even farther. If you do find it works with 2.1.2.5, then you'll want to locate a commit from `bugfix-2.1.x` from some point in the past and start from there.

Once you find a commit in the `bugfix-2.1.x` history where the feature works, then it's time to bisect. Pick a commit from halfway between that date and today and test that to see if that commit works. After each test you know that something broke in between. So you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit and test, until you find the exact day where it broke.

If you start from a point 1024 commits in the past, it will take at most 10 tests to find the exact commit that broke it.

The best tool for the job is `git bisect`. It can get to the exact issue very quickly. But it can be tricky to use if config options changed too much between tests. Fortunately the configs have not changed too much, so it should not be too hard to go through the process.

Anyway, using `git bisect` is very straightforward. You start it up and tell it that some commit is good and another is bad:
```
git bisect start  # Start
git bisect bad     # Mark current commit as "bad"
git bisect good  a1b2c3d4  # Mark some commit as "good"
```
…And now it will do a checkout of some commit halfway between those two for testing. After you're done testing, clean up your changes then use `git bisect good` or `git bisect bad` to indicate whether your test passed or failed. This causes the next commit to be checked out for testing, until there are no more commits to test, and at that point we know the exact commit that introduced the bug. (Use `git bisect reset` to finish or quit early.)

To most simply keep your configs between tests, you can use this after each test:
```
git stash
git bisect good # or bad
git stash pop
```

This can very quickly get to the bottom of any tricky issue.

ellensp, 2025-02-20:
babystepping works in the simulator also.

thinkyhead, 2025-02-20:
> I have not enabled babystepping for other axes (X, Y)

You may want to double-check that. It was enabled in the configs you attached above.

ppmanzionna, 2025-02-21:
Hi thinkyhead!
Thanks for your time. 
I tried version 2.1.2.5 of Marlin, babystepping works well.
I have Ububtu SO with Visual Studio Code for compile Marlin and i don't understand the steps to use bisect.

I'll do some tests hoping to be able to use bisect.

Thanks again.

closed 2025-06-06 by github-actions[bot]