MarlinFirmware/Marlin #27680: [BUG] Incorrect offset applied to Bed level mesh when using a fixed Home Switch/Z stop and level probe.

Maker-Paul, 2025-02-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using a fixed stop for homing connected to default Z min and a probe connected to the default separate probe connector.

The bed level mesh is created relative to the fixed Z0 and therefore an unwanted offset between the fixed Z0 and the probe Z0 is added to the mesh values. This offset can be compensated for by setting a probe Z offset, but this offset is inappropriate for homing and therefore has to be removed for subsequent homing.

 

### Bug Timeline

Not sure

### Expected behavior

The probe should be used to establish the Probe Z0 reference for the purpose of creating the mesh points (the same as when only a probe exists and has been used to home)
**Optional** When a fixed Z stop/Home switch is used a probe offset is not needed as the nozzle offset is actually relative to the fixed stop so the menu option setting a probe Z offset could be removed and it would be more logical to enable the Z bed offset that is present in the menu when only a fixed stop is used.

Please note using just a Z stop or just using a probe both work as expected.

### Actual behavior

I have to find the offset between the fixed Z Stop/Home switch and the Probe Z0. Then enter that value as a Z probe offset before I run a bed level. store the mesh and set the probe Z offset back to 0, then home the machine as a workaround way to set it up. I have enable bed leveling after homing enabled. 
Because a Fixed Z stop/Home switch to probe Z0 is not a recognized offset the numbers can get very confusing.

### Steps to Reproduce

1. Fit a fixed Z Stop and separate Level probe each connected to their respective default connectors
2. // #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and // #define USE_PROBE_FOR_Z_HOMING are both disabled
3. set the fixed Z stop switch physically to match the nozzle to bed paper pinch test. (this will mean the Fixed Z stop 0 to Probe Z 0 offset will be the same as the Z probe to nozzle offset or as near as)
4. If you run a bed level at this point and inspect the values you will see they are large.
5. run a Z offset wizard or enter the Z offset and then run a bed level. 
6. save the results
7. set the probe Z offset back to 0 and save settings (No offset is needed with a fixed Home Z0 switch)
8. Home the machine. Enable the stored bed level mesh if not set to automatically restore it.

### Version of Marlin Firmware

Latest Bugfix 2.1.x

### Printer model

Tronxy XY2 Pro

### Electronics

Stock

### LCD/Controller

Stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/18725013/Configuration.zip)

Maker-Paul, 2025-02-11:
> Maker-Paul,
>
> Why are you using a mechanical limit switch for the Z-axis when your
> Z-probe can perform the same function?

Becouse a proximity probe though good for creating a level mesh  varies
with environmental changes and takes up to 30 minutes to stabalize after
switch on (as it warms up). So its good for creating a level mesh but not
good for setting a Z zero reference. A fixed Z stop will give a perfectly
stable and repeatable Z zero reference but doese nothing else. Horses for
courses. As the machine can support both I prefer to have a deficated
sensor doing the job its best at and not having to accept a compromise. I
can use my machine day after day simply by telling it to print without
having to redo Z offsets. In fact I won't have to enter a Z offset at all
(if the bug is fixed).

borland1, 2025-02-12:
Glad you got it working. 

Mechanical limit switches are generally less accurate than a capacitive or inductive probes, however the homing accuracy is more limited by a combination of the lead screw thread pitch and your Z-axis stepper motor's micro-stepping resolution (1/16 stepping vs. 1/256 stepping).

It's easier to adjust the Z-offset, than a mechanical Z-axis limit switch, when adjusting the print's first layer squish.   You might run some test prints using a first layer mesh test print, to see if you have your mesh properly dialed in.  I am more familiar with Cura Slicer, but  Prusa Slicer might also have similar built-in tools for generating single layer mesh test print files.

Since you XY2 Pro is a bed slinger, you might want consider replacing the factory bed springs, with similar, but thicker springs (higher spring constant).  The weak stock springs tend to loosen up from constant vibrations as print time accumulates.  Also smaller diameter thumb screws are less likely to get accidentally bumped out of any proper adjustment.

Maker-Paul, 2025-02-12:
> Glad you got it working.
>
> Mechanical limit switches are generally less accurate than a capacitive or inductive probes, however the homing accuracy is more limited by a combination of the lead screw [...]

Hi Borland1. I do feel a bit  like I'm having to defend this. I have to disagree with almost every assertion you have made but I'm glad at the same time that your interested.

Even a cheap mechanical limit switch can certainly achieve a repeatability better that 0.01mm. Its the repeatability that matters in this application because what we are interested in is the relative position to other components in this case the nozzle. So it doesn't matter whether the measurement is mm, inches, standard cabbages  or a piece of paper. AS I have already said my switch is mounted on an adjustable mount that lets me adjust its position relative to the bed just by turning a thumb wheel with a resolution of 0.5mm per revolution. It takes a minute to literally dial in the Home position using the paper pinch test. Once it is set it stays set day in day out. Of course I could use a higher quality switch or optical trigger that would be highly accurate, is easily available and also cheap but I haven't found that to be necessary. AS for the mechanical accuracy of lead screws and micro stepping resolution, these are both outside the remit of this modification or the bugfix and have been addressed separately on my machines.

A proximity sensor which may have a higher resolution, which is what I think your confusing with accuracy., but has appalling long term repeatability. My proximity sensors trigger point varies by up to 2mm depending on temperature. That doesn't matter for comparative measurements where all the reading are gathered within a short enough time that the environment cant have significantly changed. A proximity sensor will only give accurate repeatability once it has settled but not during the first half hour of being switched on and not from day to day.. So creating a mesh is fine, but will my home position be the same tomorrow morning as it is at 4oc in the afternoon. I can guarantee it won't be. That is why so many people constantly have to reset the Z offset every time before they start a new print. As my firmware modification/bug fix means the probe to bed offset is measured as the first thing a bed level routine does the offset is applied to he mesh with the same accuracy as the mesh measurements themselves. the result is that my machines home position on the status screen with the mesh applied is Z 0. The produced mesh has values both slightly above and slightly below zero with a range of about 0.2 mm as would be expected from a PEI sheet on a magnetic base with a warped hot plate. The fix doesn't change the relative values of the mesh points to each other as the same value is applied to all points. All it does is move the mesh down to the zero plane.

It is not easier to drill down menu options to adjust a  Z offset than it is to move a thumb wheel by 1 notch. especially when having the homing switch set properly eliminates the need to ever have to set a Z offset either with the wizard or any other method allowing the menu options to be simplified.

I have of course carried out many first layer mesh test prints. I am now at the point where I can deliberately adjust my bed level, set the home position, run a bed level and do a  mesh test print without any further adjustments and its perfect every time, even with the bed badly out of tram.

I have three XY2 Pros all highly modified including bed springs. But this machine is as easy to use as the most expensive consumer grade machines (Bamboo Labs) Easy to set up! Stays set! This set up is as close as it gets to true automatic bed leveling and much better than the Sunlu I have with a BL touch

Pros
Cheap readily available components, will work on any board with at least 1 unused end stop socket, easy to implement (with the bug fixed firmware). doesn't require a fast mother board. brings any machine as close as possible to the elusive promise of ABL.
cons
Not widely accepted as many people perceive a proximity sensor to be an upgrade and resist the idea of fitting a mechanical switch as being somehow a retrograde step, even though each excels in its own role.

Pros and cons of other alternatives
BL touch and clones :- Roughly the same mechanical accuracy as my solution, Not affected by environment changes, mechanically complex, prone to damage, expensive, difficult to implement as a non stock upgrade on many machines.

Bed distance sensor :- (This is the future) Should be environmentally stable but I haven't tested it. Still requires a Z offset but should stay set. compensates for state of the bed in real time so no mesh, requires a capable fast mother board, expensive at the moment, complex to implement as a non stock upgrade.

I really do feel this is the best available solution to a problem that plagues so many 3D printer users. Its affordable easy to install and simplifies the operation of existing machines for non enthusiast operators.

thinkyhead, 2025-03-01:
When a Z endstop is not perfectly aligned to the Z0 position, there is the option to use the `M206 Z` Home Offset position, or if the Z endstop is at a known fixed position relative to the bed, the `Z_MIN_POS` and/or `MANUAL_Z_HOME_POS` can be set to that known position.

There may be some shortcomings to `M206` (cc: @InsanityAutomation) and to homing and leveling in general, and we will definitely be revisiting the overall UX for setup and calibration following the release of 2.1.3. That will include a single Z offset that applies to the ABL mesh and is automatically calibrated from the mesh results to minimize distortions in the print when using Leveling Z Fade.

Try experimenting with `M206` to see whether it has the effect you are looking for, and with adjusting your `Z_MIN_POS` and/or `MANUAL_Z_HOME_POS`, and we can continue discussion here whether the existing features are sufficient. I'm not sure if we'll need the workaround exactly as proposed by #27714, but it could point to some corrections we need to make to the existing features to make them more consistent.

Maker-Paul, 2025-03-01:
Thank you Scott for taking the time to look at this.   

For clarity the bug only affects a use case where both a proximity probe and fixed stop are used together. I have carried out over 300 firmware updates testing all the available combinations of config.h setting and code modifications to confirm and "fix?" the bug. Frankly I'm shocked my main board, so far, hasn't just bricked itself.

I think there are 3 things to consider . It's a deep rabbit hole so I have tried to separate the actual bug from the other issues.

1. The Bug......  I am as certain as I can be that the original programmer's didn't intend the difference between the Z stop position and the trigger height of the probe to be added on to all the mesh values. As you pointed out in an ideal machine the Z stop would align with Z zero but in practice it may well have its own offset to Z zero, making that difference even more complicated to understand. Using any of the available config options to try and compensate also affects the true home position set by the fixed stop, producing Z values on the status screen that are very difficult to understand. The only solution that worked was to either quantify that difference and take it off or not add that erroneous difference in the first place effectively making G29 work as if there is only a probe used for both homing and creating a level mesh. In the end I found the simplest solution was to not add it on in the first place, as that doesn't add any further distortion caused by floating point rounding when calculating values to compensate.
2.   The UX.....  There are as you have said inconsistencies with the Z offset being referred to as "Probe Z offset" and "Bed Z" in different menu's options. I presume that this was to try to include some descriptive reference to the device used for homing in each case. Most users will use the term "The Z offset" as the only Z offset, for the purposes of printing, is the distance between the nozzle and the Z zero plane (Bed). For the User interface all menu options could be called either Home Z offset or Nozzle Z offset. I used the predefined title "Home Z offset" in my alternative menu option because it is generic.
3.   The Code.... This is where the rabbit hole gets deep. I found that apart from the user interface using different titles for Z offset setting. The code I looked at uses at least four different named variables to apply offset values across the different leveling modes. I think from looking at the code there are nuances between different parts that indicate the code may have been written by different people at different times with subtle differences in approach. This adds massive complexity to any code alteration. For someone on the outside like me, who also doesn't have the historical familiarity with the code it makes it very difficult to reverse engineer. 

I would love to help if I can and if you think it would be of use. But can I ask you, with your lead developer hat on,  is there any written specification or guides for how specific parts of the code is meant to work or any partnering scheme where I can ask someone who knows answers to  relevant questions about the internal function of the code. I followed your previous advice to use discord and that was where someone suggested I should create my own fork and submit a pull request. So this is the first time I have cloned a repository and used git (another big learning curve) so I apologize if I have made any mistakes. Unfortunately I found that on Discord and worse Facebook I can't tell if comment is coming from a well meaning end user with an opinion, or a developer with intimate knowledge. Discord and Facebook both invite well meaning end users to  comment and start a debate about the virtues of the Bl-touch.
    I'm asking this because for example. In principle I can't think of any good reason to store anything but the bed variations in the level mesh. So I would expect the mesh to only have values slightly above or below the mesh Z zero. Then I thought maybe its done because doing several calculations to compensate for Z offset, Z zero position or mesh variations inside the main loop would have disastrous timing consequences gobbling up machine cycles. So adding all the once only values together before printing starts and doing only one calculation in the main loop would make sense.. If that is what's happening then the mesh is doubling up as a sort of buffer for other values. A question like "Is that how the code is meant to work" is a question that's just to deep for discord or Facebook.

Once again Scott thank you for all you do. I have no idea how you can keep on top of something so complex that covers such a diverse range of machines.

Regards Paul

thinkyhead, 2025-03-02:
I'm not sure of anyone's made a video on all the details of homing and leveling and probe calibration, so we just might have to make it ourselves. In the meantime I'll just go over the theory and basics here to make sure all the bases are covered and we have a map of how to proceed. Maybe we can feed this to Copilot to get some helpful hallucinations….

In the most basic form, when the Z endstop is not at exactly 0 we have offsets to specify what the homed Z position actually is. Endstops are notoriously imprecise compared to probes, and since bed probing is also in relation to the probe, not using the probe for homing is an invitation to cumulative errors, especially in setups that require periodic tramming, and those with beds that expand and change shape significantly when heated.

Of course, we want to make it possible for the corrected Z home position to be set for the Z endstop so that Z0 aligns as closely as possible with with the "most representative height" of the bed surface. This is generally the Z0 as measured in the center of the bed, though this varies from bed to bed.

And of course we also want to allow for a generally-adjustable `M206 Z` that works for a variety of common cases and does the "right thing" as it is changed. It is meant to be the lowest level adustment at the hardware level to account for the off-position of the Z endstop and nothing more.

You should find after `G28` that you can move the nozzle over to the center of the bed and then down to Z0 and it should just touch the bed. Is that is the case?
 - If so then everything is set, and now you just need to calibrate your `M851 Z` value.
 - If not, then we need to look at `G28` before we look at anything else.

To calibrate `M851 Z` you should heat up the bed and use the **Z Probe Offset Wizard** for best results. This ensures that the probe offset is at its most accurate. You may also want to run the X Axis Twist Compensation if there is any chance the X gantry might be off, or if the probe might not be aligned perpendicular to the bed throughout the probing process. Eliminating these mechanical factors is essential to getting a smallest-magnitude mesh centered closely around zero. Once we have a global Z offset added to the ABL Bilinear mesh the magnitudes will be minimized automatically.

---

Let's get a log of what you see in the course of doing the stuff above so we can analyze the output…

- Download Marlin `bugfix-2.1.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your default Configurations are applied.
- Test Procedure:
	- Reboot the machine.
	- Issue the command `M111 S247` to enable maximum logging.
	- Perform a `G28` to do your standard homing procedure.
	- Do a `G29` to probe the bed. This will also enable bed leveling.
	- Do Tramming, Probe Z Offset calibration, etc., save with `M500`, and
      repeat these steps so we can compare results before and after.
- Completion:
	- Do any other moves or commands needed to reveal problems. Take notes.
	- Copy the log output into a **.TXT file** and attach it to your next reply.

From the log file we should hopefully get a better idea of what's going on with our Z offsets and endstops maths, we can compare with the solution in your PR, and hopefully fix a fundamental aspect of the machine geometry.

Maker-Paul, 2025-03-03:
Hi Scot please find attached  test results.

[Notes and overview.txt](https://github.com/user-attachments/files/19058966/Notes.and.overview.txt)

[Z-stop_and_and_probe-no-fix-Debug.txt](https://github.com/user-attachments/files/19059008/Z-stop_and_and_probe-no-fix-Debug.txt)

Maker-Paul, 2025-03-13:
Hi Scott. I have been looking at taking this to the next level. The Anycubic Kobra 3 user interface offers only one option to "Auto Level" and their advertising claims simplicity of operation as a major plus. I have started testing under bed strain gauges the idea is to use the nozzle to probe the bed in one location where a strain gauge can easily sense contact (30grams of pressure) to establish Z Home (Z zero). A sequence of preheat nozzle and bed, wipe nozzle, home and level. would be possible to set up with no user setting up required. This would match the Anycubic set up. It does require that a separate Z stop and probe work properly together. This is the direction I think consumer machines are going now with the Anycubic, Bamboo Labs and Creality machines leading the way. It would be great if Marlin can offer similar levels of user experience even on older machines with a cheap and easy to follow upgrade path.

Nuck-TH, 2025-03-14:
Strain gauge is its own can of worms(look how "well" it fares on E3v3SE for one, also it requires hard bed surface and nozzle cleaner as well), nor it should be requirement for solving this issue. Adding support for that may be nice, but it is completely separate feature.

thinkyhead, 2025-03-16:
We provide a bunch of code that vendors are free to tailor to their needs, and some have done better than others.

The current project emphasis is on retaining current behaviors and features as-documented until the release of 2.1.3. But after the release of 2.1.3 we will continue with our various refactor and UX projects already underway. One of the things we broadly want to do is provide a good first-time setup wizard that can be tailored for different types of machines and probes so that all the important parameters will be well tuned before the first print. Some vendors, like JellyBox, have already done some work to provide first-time setup, and I'm sure we can provide a pretty good generic setup wizard that users and vendors will be able tailor to their needs following that example.

thinkyhead, 2025-03-16:
Examining your logs, the mesh results seem to be off due to some basic mis-configuration. The mesh values are all larger than +1.4mm (too high) which indicates that the `M851 Z` value is off by around that distance and/or the Z0 (`Z_MANUAL_HOME_POS` or `M206 Z` home offset) is off by about that amount.

Your logs show an `M851 Z` value set to 0 indicating the nozzle is at the same level as the probe. The probe trigger point really should be below the nozzle by some amount to prevent the nozzle itself pushing on the bed when the probe is taking readings.

Calibration:

If you find that after homing your Z0 is exact, and places the nozzle right on the bed, then your home position and/or Z home offset are set correctly. If Z0 is not right on the bed then some adjustment to the home positon is needed so you have a good Z0 to start from.

If you cannot use the probe wizard due to the wrong type of controller, then you can take height readings near the center of the bed using the `M30` command. Make adjustments to the `M851 Z` value until you get back something very close to 0, then save with `M500`.

Ultimately you want a mesh that has values as close to 0 as you can get, and this will ensure the least distortion due to the fixed part of the offset overly expressing over the Z Fade distance.

The mean value reported by `M420 C` (Mesh Centering) can also be taken as a correction to the probe Z offset.

Maker-Paul, 2025-03-16:
Hi Scott. The mesh results are off by the difference between the Z end stop switch and the trigger height of the probe which is 1.4mm difference.

Z offset value is set to zero! Because it is the Z switch which is used to home not the probe. If I set the probe offset to cancel out the 1.4mm in the mesh. The machine will crash the bed when I tell it to print because it will try to go 1.4mm below the Z end stop switch.

with Z offset set to zero my nozzle does touch the bed (Paper nip test is good).

I can use the Z offset wizard, but the 1.4mm offset will crash the bed when I print unless I remove the offset after leveling and before homing. because I'm not using the probe to home.

If you look at the comparisons in the notes and overview.txt file you will see I get a proper mesh when the machine is configured for Probe only and not with probe and Z stop enabled. That is exactly the bug! you will also see a normal mesh is created with my fix applied and both the Z stop switch and probe enabled.

Note The machine was not physically adjusted in any way between the configuration changes so the comparisons can be made. I now have two machines running this setup both behave the same.

Please review the notes and oerview.txt which I thought explained it well.

Maker-Paul, 2025-03-16:
To sum it up. The probe should only map the bed. relative to itself at  the home X and Y. 

 The Z offset should only be applied to Homing. (only needed if the homing device triggers with the nozzle above the bed.)

Maker-Paul, 2025-03-16:
https://www.facebook.com/share/v/168YEwX9DX/

With the Fix applied video shows from power on.
1. Home (double tap) Blue light on Z stop switch to left of frame activates.
2. Probe at home position to establish Mesh Z zero (Probe triggers about 1.4mm above the Z stop which doesn't trigger)
3. level 

This creates a normal level mesh.

12oclocker, 2025-04-28:
Maker-Paul is correct, this is a bug, AnyCubic i3 Mega uses a Dual Z motor and dual Z endstop switches, and if you install a BLTOUCH and use ENABLE_LEVELING_AFTER_G28, the MESH does NOT work, printing anything causes the print head to Float above the build plate by the distance of the ProbeOffset, viewing the MESH in shows values which are much too large. You have to manually edit each mesh value to pull the mesh closer to the bed.

12oclocker, 2025-07-28:
Bug: Confirmed
This bug still exist.

Z Probe Offset does NOT work if BLTOUCH and Z_MULTI_ENDSTOPS and AUTO_BED_LEVELING_BILINEAR is enabled
(Z Probe Offset has no effect)

Z Probe Offset is WORKS if BLTOUCH and Z_MULTI_ENDSTOPS and AUTO_BED_LEVELING_UBL is enabled
(Z Probe Offset works)

AUTO_BED_LEVELING_BILINEAR is BROKEN when using BLTOUCH and Z_MULTI_ENDSTOPS; only AUTO_BED_LEVELING_UBL works correctly with BLTOUCH and Z_MULTI_ENDSTOPS.

Someone please investigate and fix.

Maker-Paul, 2025-07-28:
I am exasperated by this. I have done all I can to test and comply with
thinkyheads requests. I have even provided corrected code examples and am
running my corrected version on three machines. All my machine have a Z
zero sensor and a separate proximity sensor to create the mesh.

The problem is very simple. The mesh should have its zero datum determined
using the same sensor that is used to create the mesh. That way the mesh
will only ever contain variations from the datum.

It doesn't have to be and should not be assumed the same sensor is used to
determin the actual Z zero (the bed surface)

Whatever sensor is used to determin the Z zero (the bed surface) may or may
not require using a Z offset. Any Z offset applied should only be applied
to the Z zero sensor and not to the sensor used to create the mesh.

Unfortunately the first few lines of code in the G29(create mesh) routine
calls a G28(home) to determin the datum for the mesh this assumes the same
sensor is used for both tasks establishing an incorrect datum for the mesh
sensor when seperate sensors are being used.

Unfortunatly, and I am going to say it, there appears to be a reluctance
for Marlin developers to accept that there may be a  misunderstanding of
how things should work (a programmers logic failure) as I totally agree
there is not an error in the code causing it to fail.

To avoid any further unwanted debate there are many good reasons to use
seperate sensors. For example sensorles homeing while not wanting the
nozzle to contact the bed all over the build plate to create the mesh, so
use a seperate proximity sensor for the mesh. The configuration of seperate
sensors can be chosen in the config file and should work properly. The code
doesn't have a bug in it, its just wrong.

I have dedicated many hours to reverse engineer the code track down the
problem and even submit suggested fixes. More time compileing clean
versions of firmware many installs and tests. running comparative tests of
virgin and modified firmware with many configuration changes on three
seperate machine at Scotts request, just to try and convince Marlin
Developers of what I know is the case. To be met with apparent disbelief
that there could be such a mistake.

 I have working versions of Marlin for my machines and can't give any more
time to it. I am frankly disapointed. As a developer of bespoke software
from the 1980's to this day I have had to accept that programmers make
mistakes and misunderstand functionality. They are the worst people to test
thier own work becouse they only try out what they intended the code to do
and they are often over defensive when a problem is found (Horizon! As a
postmaster) but I really didn't expect to find it in open source code.
Naively I thought open sorce developers would embrace and thoroughly
investigate all reports especially when the proof and possible solutions
are provided.

Paul Stockwell

12oclocker, 2025-07-30:
To add to Pauls response; the AnyCubic i3 Mega requires the use of the Z homing sensors even if you installed a BLTOUCH, I have 6 of those machines, and none of them can use BLTOUCH for dual Z motor homing, it fails because accuracy is never achieved using the BLTOUCH for Z homing, no matter what firmware variables you tweak. Anyone asking users to just use BLTOUCH for homing is not a solution for many multi-Z motor machines.

When using Z_MULTI_ENDSTOPS and using BLTOUCH;
AUTO_BED_LEVELING_UBL works correctly.
AUTO_BED_LEVELING_BILINEAR does not. (bug somewhere in AUTO_BED_LEVELING_BILINEAR code)
Someone needs to investigate why AUTO_BED_LEVELING_UBL works and AUTO_BED_LEVELING_BILINEAR does not.

Currently the only solution is to use AUTO_BED_LEVELING_UBL if using Z_MULTI_ENDSTOPS and BLTOUCH, since that is the only bed leveling that works with Z_MULTI_ENDSTOPS and BLTOUCH in the firmware's current state.

12oclocker, 2025-09-03:
I coded the fix, and have submitted a pull request, hopefully it is added to the marlin source code. https://github.com/MarlinFirmware/Marlin/pull/28037

12oclocker, 2025-09-04:
Yes.
During Z axis homing, Z Endstop homing occurs first, then BLTOUCH homing occurs.

AUTO_BED_LEVELING_BILINEAR will finally work with Z_MULTI_ENDSTOPS and BLTOUCH. And "Z Probe Offset" adjustments take effect immediately, you can adjust "Z Probe Offset", then home the Z axis and the new "Z Probe Offset" is in effect.

Maker-Paul, 2025-09-04:
Sounds good but Im not sure why you need a Z probe offset at all. If the
(Bltouch/probe) is used to find the datum for the bed level mesh. The mesh
should only contain the variations in the bed. So any Z offset would only
be needed to compensate for the difference between the fixed stop and the
bed surface, so wouldn't that be a Z end stop offset. Or am I
misunderstanding?

Nuck-TH, 2025-09-04:
> Sounds good but Im not sure why you need a Z probe offset at all.
Because you want to home with endstop, but you don't want leveling active all the time. Without two offsets(endstop--nozzle_on_bed) and (probe-nozzle) you can't have it. Probe offset is used only when probing, endstop offset moves whole mesh in addition to signifying where Z 0 is.

Another matter is that endstop offset can be calculated if probe offset is correct. Imo it shouldn't be done every time, it is better as on-demand function like creating mesh, because it won't change unless you mess with hardware.

Maker-Paul, 2025-09-04:
I do understand! I corrected the code submitted suggested correction to
thinkyhead and have compiled my own version which works perfectly.

I agree with everything you have said bar one thing. The level mesh should
be just a table of variations in the bed plane. So as long as the same
sensor is used to establish the datum (zero reference point for creating
the mesh) there is no probe offset.

Where the mesh is applied is determined by the fixed stop (in an ideal
world the fixed stop is at the physical bed zero with the nozzle touching
the bed) however a nozzle to fixed stop offset can make this easier unless
you have a fixed stop with a vernier adjustment (I do).

I have three machines that have been running with no offsets and have not
required recalibration since I altered the code. I get perfect first layer
every time. If I want more adhesion for a small footprint item I can lower
the fixed stop (each click is 100th of a milimeter) without having to run
any recalibrations. The machine homes before printing thus applying the
mesh at the correct height.

My fix simply removes the call  to G28 (Home to fixed stop) at the start of
the G29 (create mesh) and replaces it with temporary home with probe. Thus
removing the unwanted interaction with the fixed stop while creating the
mesh and guaranteeing the mesh contains no offset. Bonus all the numbers
reported on the display make sense.

Maker-Paul, 2025-09-05:
> Maker-Paul,
> I have updated the fix to include all probes, not just bltouch... so the
> below fix will work for your machine that uses inductive sensor. can you
> please try it and let me know.
>
>
> https://github.com/12oclocker/Marlin/tree/bugfix-2.1.x-ZEndStopWithBltouchFix

Ok I will look at it. I do appreciate your efforts. The only reason I
didn't do a pull request was becouse my fix was limited to what I use and I
didn't apply the fix or even look at the other bed leveling modes. As it
was it took me months to unscramble the code. I found the code is rather
random in approach. To me it makes no sense to call G28 at the start of the
G29 routine when there is no reason to expect G28 to be using the same
sensor that G29 uses. But it is typical throughout the code and means there
could be many unexpected behavours. Thats why when something is changed it
often causes other things to break. Its also evident that code is often
being added to fix issues instead of reviewing whether the origonal
approach should be reviewed and rewriting a simpler routine.

I was responsable for systems analysis on several large projects back in
the day. Does it show 🤣🤣.

I did ask scott for desighn guides or contact with a Marlin developer. But
nothing was forthcomming.

12oclocker, 2025-09-05:
Maker-Paul,
the problem of probes not working with Z endstops has been an issue for a long time in marlin, especially if you have multiple Z end-stops with multiple Z axis. pull request #28037 is aimed to fix those issues for all machines. I have a lot of machines using Z_MULTI_ENDSTOPS and Probes, like you I had very little time to actually examine the source code and solve the issue, everything I had done for years was just workarounds, but I finally had some time to tackle it and pull request #28037 is the result. It is designed to work for all machines that use Z endstops with probes. Their is a lot of macro voodoo in marlin, so I made sure that all the code changes mimic the original author style of code, and even uses their TERN and TERN0 macro's to reduce compiled code space. Z_HOMING_WITH_PROBE_AFTER_Z_ENDSTOP will need enabled in Configuration.h to enable the new functionality to take effect.

thinkyhead, 2025-09-27:
Well, I have been busy working on other priority issues and taking a sanity break, but now revisiting bugs and issues that need addressing. I appreciate your frustration at my daftness, but god knows we spent many tedious weeks going over the endstop offsets, probe offsets, workspace offsets, etc., and it is never fun to revisit this area of the code. We fix something for one user and end up breaking something else. So I want to be very, very sure if we need some other special sauce beyond the various settings we already provide, and I am admittedly quite reluctant to add something new.

I have had a difficult time understanding the crux of the problem, perhaps because of the verbosity surrounding it. But after a re-read of the conversation it seems the core issue is simply an unreliable probe whose trigger-point-to-nozzle distance cannot be known to any degree of accuracy.

So the obvious question is: Can we automatically determine the correct `probe.offset.z` to apply during `G29`?

Yes.

The simple solution is to do an initial probe to establish the Probe Z Offset so we can determine what it should be based on the `Z0` that was established by the Z endstop:

1. Home with Z Endstop. If you were to move the nozzle to the center of the bed and go to Z0 at this point, the nozzle would just touch the bed.
2. Move the Probe to Bed Center and probe down until it triggers, and immediately set `probe.offset.z = -current_position.z`.
3. Now proceed to probe the bed using the established height of the probe trigger point.

It seems to me that this should do the trick. Do you see any flaws in that plan?

Maker-Paul, 2025-09-27:
Thank you Scott for revisiting this. To be clear I never suggested that my
fix is the best or only solution just that it demonstrated a possible
solution to a limitation in the way G29 works.

My fix simply established or rather removes the Z offset by probing with
the probe at the XY home cordinate. I set the Z stepper to zero and then
probe the bed mesh. The mesh then holds only the variations from the probe
trigger at XY home coordinate. I then home using the Z stop which resets
the Z stepper zero.

Everything else then works as expected. Including any Z home offset between
the Z stop and nozzle.

This solution allows any device to be used for Z homeing including peizo,
sensorless, lidar or whatever one wants. Whether or not such a sensor still
requires a Z offset or not.

The reason to probe at the XY home coordinate to establish stepper zero
before G29 is so that I can use a feeler gauge or paper nip test to set my
Z endstop with the nozzle at the same coordinate. That way the mesh zero
point and the actual bed Z0+any z offset are syncronised.

On my machines I eliminated the need for a Z home offset by having a Z stop
that is mechanically adjustable with a resolution of 0.005mm. Therefore I
no longer need any Z offsets and the machines are 100% stable. Very cheap
and easy to impliment. (Even on old kit with a standard Z stop microswitch
and some printed parts). Cheap and easy upgrade paths are always popular.

If I have to change a nozzle for example. I set the home possition with a
feeler gauge and auto level (G29) and that's it.

Thanks Paul

Maker-Paul, 2025-09-27:
Youve got it scott😀😀😀😀. I zero the stepper and then probe instead of
measuring the probe height and taking it off the bed mesh measurements.
Becouse I found calculation introduced a 0.01 rounding error that showed up
on the status display as z 0.01mm when the nozzle was moved down back down
to zero after homeing.

hberntsen, 2025-09-29:
Thanks Scott for revisiting! I feel that this indeed is a complex subject to get right and I'm thankful for your efforts.

> Do you see any flaws in that plan?

At some point we need to set the Z0 home offsets for a new printer. In the menu on the printer we have an option for it: the one that triggers "M428 home offsets here". To do that on my printer, I need to go to the front left of the bed to do so since that is where X0Y0 is. In your plan you move the probe to the bed centre, which is a different place from where I calibrated the home offset, and thus won't correctly read the probe offset; that should be done at the same place where I manually set the Z0 home offset.

My current flow for setting up my printer is:
1. Move the head to where X0 Y0 should be, Z somewhere close to 0 (does not matter that much)
2. M428 home offsets here.
3. Run the probe measure wizard in the centre to measure the probe height (with [this bugfix](https://github.com/hberntsen/Marlin/commit/1176d5c23dbf84cc8f0d8738ba81915b694fc488)
4. Run bed levelling and use the Z offset it finds as Z home offset: https://github.com/hberntsen/Marlin/commit/e9c317999c8cc059077781d8893b9caf2b388575

As far as I know, bed levelling does not save its `plane_equation_coefficients.d` to EEPROM but it does represent the correct Z home offset. My probe needs to manually be deployed and we don't want to do that at the start of every print or printer reboot. I save the coefficient every time bed levelling is run. I guess that makes sense to do. For example, when changing the build plate to a different glass plate or so I only need to re-run bed levelling and all is good.

Note that I'm also the author of #28025, which I resolved with github.com/hberntsen/Marlin/commit/89234afbb22463593e4c1d00a66545b1f3ac7399 and I'm also using that.

hberntsen, 2025-09-29:
Sorry, English is not my native language. The nozzle is approx at Z=5 when it is at its Z home position. So it is higher than the bed, not below. In the home position it is not above the bed though because it is a bit more to the left and in front of the print bed (so X and Y). Filament would fall next to the print bed in the home position. At that position the probe won't hit the bed.

I volunteer at a project where children around 10-15 years old learn working with wood, nuts/bolts etc, electronics and CAD by building their own 3D printer. So we have to work with a few centimeters of tolerance.

thinkyhead, 2025-09-29:
> The nozzle is approx at Z=5 when it is at its Z home position

If your optical endstop is mounted 5mm above Z0, your `Z_HOME_POS` should be 5.0 (or whatever value causes a move to Z0 to put the nozzle at the bed surface).

hberntsen, 2025-09-30:
Yeah, makes sense now! I should have taken a bit more time to read through everything you already posted in your PR but I was also squishing replying here in between my duties at work.

Seems like we have quite similar but still different use cases for working with both Z endstop switches and a probe. I'll leave it up to Scott to now see what's best in terms of Marlin code to do :).

Maker-Paul, 2026-04-15:
Becouse the issue has not been resolved or  rejected in over a year and the pull request has just been allowed to die. I can't keep spending my time trying to manage a branch just to fix something that is being ignored.

It is such an easy fix based on Scott's own suggestion that I'm hoping by asking for it as a feature Scott will impliment it. Then I won't have to keep downloading the latest bugfix and having to modify it every time.

I want to continue to develope cheap easy upgrades for a wide veriaty of machines. Improving day to day stability is fundemental to that end. I am working on an easy three point bed mounting method with peizo sensing at the reference point. That would eliminate any need for a user setting of a Z offset so even an old bed slinger can have a single calibrate option.

If the feature request is granted I will take down the branch.