MarlinFirmware/Marlin #27683: G30 - unwanted bed movements

mafr95, 2025-02-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Each time the G30 command is executed, the probe moves to the XY coordinates of the nozzle before probing and return after. I understand the idea that the Z probe should start exactly at the xy-position of the nozzle. However, this behavior causes my bed to constantly move between measurements with L-Corner or the octoprint plugin AutoBim, which makes it impossible to adjust the bed screws.

[https://www.youtube.com/watch?v=Bxp1_7POnQ4](url)

The probe did not move in the earlier Marlin-2.1.2.4 release.

### Expected behavior

No movement expected before and after probing.

### Version of Marlin Firmware

Marlin bugfix-2.1.x

### Printer model

Sidewinder X1

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/user-attachments/files/18766574/Configs.zip)

ellensp, 2025-02-12:
To adjust screws you use  ASSISTED_TRAMMING   G35

G30 tells you on serial console X,Y and Z position  eg "Bed X:20.00 Y:20.00 Z:0.000"

ellensp, 2025-02-12:
22 months ago a commit entitled "Fix G30 behavior" added a move after probing to move the hotend to the probe point

This looks to be deliberate and intentional.

https://github.com/MarlinFirmware/Marlin/commit/25e5a3597fed4c9c9c0b074e4aa7c5a7c69d0403

"do_blocking_move_to(old_pos);"

mafr95, 2025-02-12:
> To adjust screws you use ASSISTED_TRAMMING G35

G35 works fine, but plugins in octoprint like AutoBim use other commands including G30. The L-Corner option in my BTT TFT 35 also seems to use G30.

Therefore, it would be helpful to have an option in Marlin to disable the 

> “do_blocking_move_to(old_pos);”