MarlinFirmware/Marlin #27684: [BUG] Robo 3D R1+ Marlin bugfix 2.1.x not homing Z axis

LexAKABox, 2025-02-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

sorry. im new here. please excuse improper procedures. i have thoroughly scoured this site, Marlins site and everywhere else i can find, but have not found a solution.

After successfully building the firmware VIA Visual Studio Code, when homing the Z axis it only drops about 10mm and stops.
i had to reverse the Z homing direction from the original 1.1x firmware (-1 to 1) otherwise it would try to home moving up. using the arrow commands all axis move in the proper direction. i have tried a lot of different "fixes" with no change. i made so many changes, i just reloaded the original files and am starting over from scratch. i am now back on the stock configuration.h/adv.h files from the marlin bugfix 2.1.X configuration examples only changing the temperature sensor for the E3D (5). i have attached the compiled firmware as well as the config files I'm using. any help would be greatly appreciated. i cannot tell if the auto bed leveling will work as i cannot get past homing. i recognize ABL is currently not enabled in the attached files. i wanted to get the homing working as it uses the same switches. thanks again in advance everyone!

https://github.com/user-attachments/files/18790363/firmware.zip
https://github.com/user-attachments/files/18790367/Configuration.zip
https://github.com/user-attachments/files/18790368/Configuration_adv.zip

### Bug Timeline

seems to be in all variants of the 2.x 

### Expected behavior

z axis home

### Actual behavior

z axis moves down 5-10 mm and stops regardless of where it is in the plane (after swapping the z home direction. it was moving up instead of down) in all previous builds home direction is -1 but i had to switch it to 1 in order to get it to move down.

[image]

### Steps to Reproduce

attempt to home z axis or all axis either via software or terminal. 

### Version of Marlin Firmware

bugfix 2.1.x SHORT_BUILD_VERSION "2.0.5.3"

### Printer model

Robo 3D R1+

### Electronics

stock electronics other than an E3D V6 hotend

### LCD/Controller

none

### Other add-ons

none

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Same as my slicer

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

borland1, 2025-02-17:
Have you tried debugging the hardware by connecting your printer to a serial terminal program and sending G-code to check that your z-axis limit switch is working properly?  

Sending g-code M119 will report Endstop State.

```
> M119
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED
z_probe: open
filament: open
```

[Marlin G-code M119](https://marlinfw.org/docs/gcode/M119.html#:~:text=Use%20this%20command%20to%20get%20the%20current%20state,runout%20sensors%20are%20also%20reported%20with%20this%20command.)

[Endstops Reference](https://marlinfw.org/docs/hardware/endstops.html)

LexAKABox, 2025-02-17:
i have not tried that. but it works properly with the original Marlin 1.0
firmware. i will test that when i get back into town.

borland1, 2025-02-17:
OK, There's a Endstop Threashold Setting In the Configuration.h file, at the end of the "**Endstop Settings**" section.  Also, a check for stuck or broken endstops.

```
 /**
 * Endstop Noise Threshold
 *
 * Enable if your probe or endstops falsely trigger due to noise.
 *
 * - Higher values may affect repeatability or accuracy of some bed probes.
 * - To fix noise install a 100nF ceramic capacitor in parallel with the switch.
 * - This feature is not required for common micro-switches mounted on PCBs
 *   based on the Makerbot design, which already have the 100nF capacitor.
 *
 * :[2,3,4,5,6,7]
 */
//#define ENDSTOP_NOISE_THRESHOLD 2

// Check for stuck or disconnected endstops during homing moves.
//#define DETECT_BROKEN_ENDSTOP
```

Might want to try independently uncommenting those #defines and see if behaviour changes.

closed 2025-06-04 by github-actions[bot]