MarlinFirmware/Marlin #27689: [BUG] pioarduino for esp32 support using platformio is broken in bugfix / Marlin 2.1.3 beta 1 / Marlin 2.1.3 beta 2, but ok for Marlin 2.1.2.5

luc-github, 2025-02-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am working on ESP3DLib V3.0 which is based on ESP3D V3.0, the code is using arduino esp32 version 3.0.4 which use espressif IDF 5.1.4

Unfortunatly since november 2023 espressif stop to support platformio:

https://github.com/platformio/platform-espressif32/issues/1225

Which mean any release after esp32 arduino 2.x is not supported by platformio neither, because no solution was found between both side, the community launched `https://github.com/pioarduino`, and especialy : `https://github.com/pioarduino/platform-espressif32` 
This repository release platformio package for every 3.x release done for esp32 arduino core

The syntax to use the pioarduino is the following:
; Arduino  core 3.0.4 - IDF 5.1.4
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip

I use it for several month now on ESP3D and wanted to update Marlin in same way, so I recently have started the porting of MArlin to esp32 3.0.4 because of several bug fixes and new features in esp32 arduino 3.x, including new chips support 

I am currently working on Marlin 2.1.2.5 porting and I have only have migrations issues due to breaking changes in API: https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html which should not be a problem 
but I can launch the compilation.

Issue description:

Unfortunatly in bugfix/Marlin 2.1.3 beta 1 / Marlin 2.1.3 beta 2 the compilation does not even start:

```
Executing task: C:\Users\luc\.platformio\penv\Scripts\platformio.exe run --environment mks_tinybee 

Processing mks_tinybee (board: marlin_MKS_TinyBee; platform: https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
'"CC"' is not recognized as an internal or external command,
operable program or batch file.
Error: Failed to parse Marlin features. See previous error messages.
============================================================================================ [FAILED] Took 1.66 seconds ============================================================================================

Environment    Status    Duration
-------------  --------  ------------
mks_tinybee    FAILED    00:00:01.655
======================================================================================= 1 failed, 0 succeeded in 00:00:01.655 =======================================================================================

 *  The terminal process "C:\Users\luc\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'mks_tinybee'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

It is look like the issue is definitly not in platformio but in Marlin prebuild code check that do seems to like the `platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip` package, but I was not able to find a solution because I am not familiar with Marlin prebuild process.

So is it possible to restore the support for pioarduino in bugfix before finale release? because futures updates will come from https://github.com/pioarduino/platform-espressif32 not platformio/platform-espressif32 package which is now frozen for arduino

### Bug Timeline

Started after 2.1.2.5

### Expected behavior

I expect compilation start like when using Marlin 2.1.2.5

### Actual behavior

Executing task: C:\Users\luc\.platformio\penv\Scripts\platformio.exe run --environment mks_tinybee 

Processing mks_tinybee (board: marlin_MKS_TinyBee; platform: https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
'"CC"' is not recognized as an internal or external command,
operable program or batch file.
Error: Failed to parse Marlin features. See previous error messages.
============================================================================================ [FAILED] Took 1.66 seconds ============================================================================================

Environment    Status    Duration
-------------  --------  ------------
mks_tinybee    FAILED    00:00:01.655
======================================================================================= 1 failed, 0 succeeded in 00:00:01.655 =======================================================================================

 *  The terminal process "C:\Users\luc\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'mks_tinybee'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

### Steps to Reproduce

just change in esp32.ini the platform for the esp32 board:
```
; Arduino  core 3.0.4 - IDF 5.1.4
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
```

### Version of Marlin Firmware

bugfix/Marlin 2.1.3 beta 1 / Marlin 2.1.3 beta 2

### Electronics

esp32 boards

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

it is prebuild part in platformio, not related to

ellensp, 2025-02-16:
Look in buildroot/share/PlatformIO/scripts/preprocessor.py

code was added to filter out unused local compilers ending with "-elf-g++" 

It looks for "xtensa32" in the path to determine if its a esp32 to allow for that compiler name to end  in "-elf-g++"

luc-github, 2025-02-16:
Thank you I will try to test on linux as I am on window and will check the filter you mentioned

luc-github, 2025-02-16:
I confirm issue is on window and not under linux like you reported 
I have tested on linuxmint and it seems ok :

[image]

I will check difference between windows and linux in `buildroot/share/PlatformIO/scripts/preprocessor.py` then

luc-github, 2025-02-16:
I can confirm the filter is the root cause:
on platformio package:
gccpath=C:\Users\luc\.platformio\packages\toolchain-xtensa32\bin\xtensa-esp32-elf-g++.exe
on pioarduino package
gccpath=C:\Users\luc\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-g++.exe 
Directory name changed from `toolchain-xtensa32` to  `toolchain-xtensa-esp32`, so changing in   `buildroot/share/PlatformIO/scripts/preprocessor.py` :
` if not gpath.stem.endswith('-elf-g++') or "xtensa32" in str(gpath):`
to 
 `if not gpath.stem.endswith('-elf-g++') or "xtensa" in str(gpath):`

allow to manage platformio and pioarduino 

I will do PR once I am done porting,  thank you @ellensp for the hint

luc-github, 2025-02-17:
PR created just in case

closed 2025-02-19 by thinkyhead