MarlinFirmware/Marlin #27690: [FR] (feature summary)

Simmoethan, 2025-02-16:
### Are you looking for hardware support?

Any chance we can get support for the Tiertime up300? 
The software and slicer are probably one of the worst that i have seen and it has little to no support.

### Describe the feature you want

Updated firmware for the touch screen

ellensp, 2025-02-16:
Closed source printer with preparatory closed hardware.
There is no where to even begin looking at this without access to the real printer and a lot of reverse engineering.

ellensp, 2025-02-16:
Main board (CPU is its own PCB module)
https://cdn.hackaday.io/images/2057051547926848445.JPG

Display Board 
https://shop.tiertime.com/wp-content/uploads/2019/03/up300-%E8%A7%A6%E6%91%B8%E5%B1%8F-%E8%BF%B7%E4%BD%A02ES.png

Simmoethan, 2025-02-16:
So what would I need to do?
I got given the printer from my school

ellensp, 2025-02-16:
This is a non realistic ask. Porting Marlin to a new undocumented controller is not trivial.

It requires someone with the printer and the electronic skills and software abilities.