MarlinFirmware/Marlin #27691: StallGuard Threshold Stall Sensivity Sensorless Homing

SP892512, 2025-02-16:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I use Marlin-2.1.2.5 with Visual Studio Code and PIO to Compile.

I have activated Sensorless Homing and defined the following regarding Stall Sensitivity with TCM2209 Motor Drivers on the SKR 3 EZ Board.

  #define SENSORLESS_HOMING // StallGuard capable drivers only

  #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY  100
    #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY  105
    #define Y2_STALL_SENSITIVITY Y_STALL_SENSITIVITY

If I do a read out M503 with ponterface I received different Values - it seems that the defined values from below where not set to the software. 

echo:; StallGuard threshold:
echo:  M914 X8 Y8

Thank you very much.
Best regards
Stefan

### Expected behavior

M914 should show the Stall Sensitivity Values defined in the Software

### Actual behavior

M914 shows default Values 8 for X and Y

### Version of Marlin Firmware

Marlin-2.1.2.5

### Printer model

Anycubic Kobra

### Electronics

SKR 3 EZ

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2025-02-16:
Marlin settings basics 101:

Settings that can be changed via gcode or LCD are saved in EEPROM
The EEPROM setting are independent of firmware settings.
Updating firmware will not update the values stored in EEPROM (1)(2)(3)
STALL_SENSITIVITY is one of these variables.

(1) On the first time you update with new firmware, if the EEPROM version does not match or the saved checksum is invalid, then the default vales in firmware are written to EEPROM to be used.
(2) If you enable the option EEPROM_INIT_NOW then each and every time you flash the firmware the EEPROM is reset to the firmware defaults.
(3) You can manually set EEPROM contents back to firmware defaults using gcode M502 - Factory Reset then M500 - Save Settings

closed 2025-02-16 by ellensp

SP892512, 2025-02-16:
> Marlin settings basics 101:
>
> Settings that can be changed via gcode or LCD are saved in EEPROM The EEPROM setting are independent of firmware settings. Updating firmware will not update the values [...]

> Marlin settings basics 101:
>
> Settings that can be changed via gcode or LCD are saved in EEPROM The EEPROM setting are independent of firmware settings. Updating firmware will not update the values [...]

Hello eelensp,

thanks for fast supply and exlpanation. Sorry for question again but what does I have to do that the Values from the Firmware are stored in the EEprom?

Does I have to enable EEPROM_INIT_NOW so that theses Values are stored?

Best regards
Stefan