MarlinFirmware/Marlin #27698: [BUG] Some symbols lost in answer

smoluks, 2025-02-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Some symbols lost in answer (when board connected to octoprint server by usb) or board can't read some symbols and do repeat command (when connected by uart) on BIGTREETECH SKR MINI E3 V3.0

All OK on BIGTREETECH SKR MINI E3 V2.0

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Feb  6 2025 18:36:44) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-5 KINEMATICS:Cartesian EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Ender-5

### Electronics

BIGTREETECH SKR MINI E3 V3.0

### LCD/Controller

Stock

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Send: N10224 G1 X111.669 Y92.731 E174.53642*85
Recv: ok
Send: N10225 G1 X112.319 Y92.802 E174.55208*86
Recv: o T:234.80 /235.00 B:104.31 /105.00 @:82 B@:127
Recv:  T:234.89 /235.00 B:104.40 /105.00 @:79 B@:127
Recv:  T:234.87 /235.00 B:104.57 /105.00 @:80 B@:127
Recv:  T:234.91 /235.00 B:104.71 /105.00 @:79 B@:127
Recv:  T:235.19 /235.00 B:104.85 /105.00 @:73 B@:127
Recv:  T:234.87 /235.00 B:104.98 /105.00 @:80 B@:127
Recv:  T:234.87 /235.00 B:104.79 /105.00 @:79 B@:0
Recv:  T:235.06 /235.00 B:104.36 /105.00 @:75 B@:0

[Configuration.zip](https://github.com/user-attachments/files/18897223/Configuration.zip)

ellensp, 2025-02-21:
The processor on that motherboard is very easy to overload

enable ENDSTOP_INTERRUPTS_FEATURE  as a start

ellensp, 2025-02-21:
increase buffers as well as increasing #define MULTISTEPPING_LIMIT 16 to 32

smoluks, 2025-02-21:
testing...

smoluks, 2025-03-09:
With
#define MAX_CMD_SIZE 96
#define BUFSIZE 4
#define TX_BUFFER_SIZE 256
#define RX_BUFFER_SIZE 256
Usb connection works fine, but uart-to-uart has a lot of:
`Send: N839 G1 X96.985 Y115.733 E27.30319*100
[...]
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Recv: Error:Line Number is not Last Line Number+1, Last Line: 838
Recv: Resend: 839
Recv: ok`

`Send: N843 G0 X96.745 Y113.752*29
[...]
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Recv: Error:checksum mismatch, Last Line: 842
Recv: Resend: 843
Recv: ok`

smoluks, 2025-03-11:
Confirm:
printing from PC via usb-uart converter bugged
printing from PC via usb - all OK

Looks like UART2 has problem on this board
[image]

ellensp, 2025-03-11:
Provided config files are a mess of reformatting, making it difficult to see what has or has not been modified.

[Cleaned Configuration.zip](https://github.com/user-attachments/files/19183032/Cleaned.Configuration.zip)
New we can see what has and has not been enabled.

I see neopixels, these are terrible at sucking cpu cycles and block interrupts (like moving data from a hardware serial buffer)

As a test disable your neopixels, see if the issue remains.

smoluks, 2025-03-12:
`Send: N200 G1 X132.954 Y101.061 E11.51369*87
Recv: Error:No Checho:busy: processing
Changing monitoring state from "Printing" to "Error"`

smoluks, 2025-03-20:
Update: the problem occurs only when printing curves, and the printing freezes
and only whet uart as source of G commands

@ellensp do you know marlin uses interrupt priorities on stm32?
Thanks

cbagwell, 2025-03-21:
If you've not done it yet, you can use Arc Welder plugins to convert curves to G2/G3 to reduce serial port usage.

As I was tuning my Ender 3 Pro with this same board and with Input Shaping and approaching 200mm/s prints, I would lose characters even on the USB uart and never found a solution.

Its been almost a year since I last looked into it but I seem to recall that with this board you need to edit `ini/stm32g0.ini` and the `-DSERIAL_RX_BUFFER_SIZE=1024 -DSERIAL_TX_BUFFER_SIZE=1024` values to really increase the buffer sizes instead of those header file values.

smoluks, 2025-03-21:
Thanks
USB uart works well with 0% resend ratio

thinkyhead, 2025-03-27:
> Its been almost a year since I last looked into it but I seem to recall that with this board you need to edit `ini/stm32g0.ini` and the `-DSERIAL_RX_BUFFER_SIZE=1024 -DSERIAL_TX_BUFFER_SIZE=1024` values to really increase the buffer sizes instead of those header file values.

These are now standard for some of those environments.
Do you see any more in that file as it is today that ought to have those flags added?

closed 2025-03-27 by thinkyhead