MarlinFirmware/Marlin #27699: [FR] (feature summary)When the endstop is triggered, it could rebounce or slide by fixed distance.

dty717, 2025-02-21:
### Describe the feature you want

There is fixed distance between real stop and endstop, so I want to fix the problems. Is that can be solve by configuration or need firmware modified.

ellensp, 2025-02-21:
Already exsists

#define HOMING_BUMP_MM      { 5, 5, 2 }       // (linear=mm, rotational=°) Backoff from endstops after first bump
#define HOMING_BUMP_DIVISOR { 2, 2, 4 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)

closed 2025-02-21 by ellensp

dty717, 2025-02-21:
Ok, i will look at it, thanks.