MarlinFirmware/Marlin #27703: [FR] ZHOME_BEFORE_LEVELING appears in only one line

mennucc, 2025-02-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

dear all,

I was browsing in the source code (I have some problems with UBL ABL), branch `origin/bugfix-2.1.x`  ; I noted that the define ZHOME_BEFORE_LEVELING appears in only one line in the whole source code , namely

> ./Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:323:    #if ALL(DWIN_LCD_PROUI, ZHOME_BEFORE_LEVELING)

this is weird

### Expected behavior

A define should appear also somewhere else, e.g. in  configuration files, with some documentation

### Version of Marlin Firmware

origin/bugfix-2.1.x

### Printer model

(not relevant)

### Electronics

(not relevant)

### LCD/Controller

(not relevant)

### Other add-ons

(not relevant)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

(not relevant)

ellensp, 2025-02-22:
added in https://github.com/MarlinFirmware/Marlin/pull/26003

is DWIN ProUI specific defines

bmourit, 2025-02-26:
Was it added in that commit, or just used? I only see the define being used there. Maybe unrelated, but I have noticed new build no long Z home before leveling for me either, and no config changes.

mennucc, 2025-05-30:
the situation is the same as when I posted this issue

bmourit, 2025-05-30:
Are you experiencing a bug? Or what problems are you having with this? Any issues with your bed leveling? I'm not sure I understand what the issue is. Are you confused about the wording not being specific to DWIN ProUI?

closed 2025-09-14 by github-actions[bot]