MarlinFirmware/Marlin #27713: [FR] ESP32 over wifi, slicer control

Ammar56788, 2025-02-26:
### Is your feature request related to a problem? Please describe.

Hello;

i after closing my [preview issue](https://github.com/MarlinFirmware/Marlin/issues/27706) as completed, searched in many forum but i dont found any thing about it,
then i decided to ask my question here again to the developers.

I have esp32 as main board controller for 3dprinter, and I enabled esp3d.
I can use browser to control 3d printer, and upload gcode etc...

But if I want to connect it with slicer over wifi, and send data with wifi gcode, I can't.
I think maybe there are disabled feature that I don't know, or maybe I must use wifi support instead of esp3d.

Second, I don't know esp3d and wifisupport difference..

### Describe the feature you want

controllimg ESP32 over wifi, with slicer.(using ip or url)

### Additional context

i dont find answer in many forum

Ammar56788, 2025-02-26:
Also when processing the gcode on the SD card by esp32 I can't send any command by esp3d, it gives an error: busy.... What to do to control the flow speed and feed rate?

ellensp, 2025-02-27:
This is not a support forum

esp3d is from a 3rd party https://github.com/luc-github/ESP3D

What your looking for is "Serial/Wifi bridge using configurable port 8888" part of esp32 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2025-02-27 by ellensp