MarlinFirmware/Marlin #27722: [BUG] Marlin 2.1.3-b2 , Ender 3, LCD screen freeze/ lag while printing

RF47, 2025-03-03:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

During printing the screen freezes and the knob does not respond, when printing is finished it works correctly again.

### Bug Timeline

This bug originated after flashing Marlin 2.1.3b2

### Version of Marlin Firmware

2.1.3-b2

### Printer model

Creality Ender 3

### Electronics

Creality v4.2.7

### LCD/Controller

stock LCD Screen

### Other add-ons

Custom thermistor 5 calibration (file attached)

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config Marlin 2.1.3b2.zip](https://github.com/user-attachments/files/19059777/Config.Marlin.2.1.3b2.zip)

Lazar0v, 2025-04-27:
It's the same with me.
GD32 MFL latest bugfix-2.1.x code 

[CrealityV422MFL-PRO.zip](https://github.com/user-attachments/files/19931928/CrealityV422MFL-PRO.zip)

classicrocker883, 2025-04-28:
so try this... do you see if the screen is frozen after it has turned off? press in the knob and the screen should unfreeze. turning the knob left or right does not fix this.

I do have a fix, I will be putting a **PR** in time. but just to confirm, is this your situation? that the LCD is frozen only after it is turned off?

Lazar0v, 2025-04-28:
Only while printing does it freeze (slowly refresh) the display.
Ender 3 GD32 chip
BOARD_CREALITY_V422_GD32_MFL

ellensp, 2025-05-03:
@classicrocker883 GD32 chips are not identical to STM32, this is why the  GD32 MFL build environment was created to address the differences.

classicrocker883, 2025-05-04:
i meant the pinout is according to the datasheet. not sure if the board layout/wiring matters if its different. and while I never tried flashing one, it would be interesting to see if it takes.

@RF47 never answered the question if this is ProUI or what LCD type. but if its GD32F103 then its definitely compatible with STM32F103.

if you look at  Creality CREALITY_CR4NS (GD32F303RET6) - Sold as "Creality Ender-3 V3 SE CR4NS200320C13 Motherboard", this uses STM32F1. so this is what I'm saying that STM32F1 ini `env:` just might work with all GD32F303 chips/boards

RF47, 2025-05-04:
I have the stock LCD screen from the Ender 3, and the MCU is an STM32F103.

closed 2025-07-06 by RF47