MarlinFirmware/Marlin #27723: [FR] Support M550 for string configuraiont

crossan007, 2025-03-04:
### Is your feature request related to a problem? Please describe.

Currently, Marlin requires the machine name to be set at compile time using the CUSTOM_MACHINE_NAME definition in Configuration.h. 

There is no G-code command available to update the machine name dynamically. 

This can be limiting for users managing multiple machines, networked printers, or wanting to adjust the name via host software or macros without recompiling and flashing the firmware.

### Are you looking for hardware support?

No, this is a firmware feature request and is not related to any specific hardware.

### Describe the feature you want

I would like Marlin to support the M550 G-code command (similar to how it works in RepRapFirmware) to dynamically set the machine's name during runtime.

Example syntax:

M550 P"NewPrinterName"

This would allow:

    Updating the machine name on the LCD (if applicable).
    Having the new name reported in M115.
    Optionally updating any network hostname (if applicable, e.g., with ESP32 or Ethernet-enabled boards).

This would provide flexibility to change the machine identity without needing to recompile the firmware.

### Additional context

Use cases for this feature:

    Dynamically identifying machines in multi-printer environments.
    Changing the printer's name via OctoPrint or other remote interfaces.
    Temporary renaming for specific jobs or users.
    Aligning with existing G-code standards, improving compatibility with slicers and ecosystems that expect M550.

The only other reference to `M550` on this repository is https://github.com/MarlinFirmware/Marlin/issues/7076#issuecomment-487988693

ellensp, 2025-03-04:
https://reprap.org/wiki/G-code#M550:_Set_Name  is mentioned. This is the unofficial list of gcode for all 3d printers

ThomasToka, 2025-03-04:
duplicate of https://github.com/MarlinFirmware/Marlin/issues/21653