MarlinFirmware/Marlin #27727: [BUG] Starting spindle causes stepper driver to reset

buergi, 2025-03-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm using Marlin (LTS-2.1.2, but its the same in the bugfix branch) for a small CNC mill with the old Bigtree SKR v1.3 board. As it is only a small mill with a small spindle motor I connected the motor directly to the SKRs heat bed output.
The problem was, that very often when I starting running gcode the travelling distances were 4x larger than configured. I noticed it only happens with gcode with running spindle. After several hours of debugging I found that suddenly spinning up the spindle draw so much power that the supply voltage dropped causing the TMC 2209 stepper drivers to reset to there default configuration with 8 instead of 32 micro steps.

### Expected behavior

I was not sure if I should post it as a bug report or a feature request, but finally I chose to consider it a bug that the M3/4 command does directly issue a set_pwm_duty command with the target PWM instead of gradually increasing it over a time of several seconds. Certainly there are spindle motors with RPM regulation which do not require it, but at least having the option to configure a ramp-up time would be great.

Maybe it is also possible to detect such a reset of the stepper drivers somehow and either reconfigure it or stop the machine.

### Actual behavior

Currently M3/4 set the spindle output directly to the target PWM.
The drop in supply voltage caused the TMC2209 to reset to its default settings and the steppers slammed into workpiece with 4x the configured movement speed.

### Steps to Reproduce

I reduced the gcode to
````
  M3 S255
  G0 X10 Y10 Z10
````
whereas something like this works
````
  M3 S150
  G4 S1
  M3 S200
  G4 S1
  M3 S255
  G0 X10 Y10 Z10
````

### Version of Marlin Firmware

LTS-2.1.2

### Printer model

Similar as https://www.thingiverse.com/thing:4251212

### Electronics

Bigtree SKR v1.3

### LCD/Controller

RAMPS 1.4 LCD12864

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[conf.zip](https://github.com/user-attachments/files/19116748/conf.zip)

DerAndere1, 2025-03-18:
I guess the expected behaviour for M3/M4 is to accelerate the spindle and wait until the final pwm value is reached. I can try to implement a config option, something like 'DEFAULT_MAX_ACCELERATION_SPINDLE' . An additional improvement would be to allow setting the spindle acceleration at run time with 'M201 R'

DerAndere1, 2025-03-25:
@buergi  I have opened pull request https://github.com/MarlinFirmware/Marlin/issues/27759. You can test the branch that contains my changes: https://github.com/DerAndere1/Marlin/tree/spindle_acceleration and let me know if it fixed your issue.

buergi, 2025-03-27:
@DerAndere1 thank you so much for your quick implementation of my request. I really hadn't expected this as there are almost 700 other open issues. I really appreciate your effort here.

I backported your changes to lts-2.1.2 for convenience. It compiled but did not work due to `ENABLED(DEFAULT_ACCELERATION_SPINDLE)` always evaluated to false for any numeric value other than 1, with 1 it did not compile.  After several changes I got it to build, for details see review of the pull request. However, after issuing a M3 the firmware completely blocks, presumably as it hangs in the while loop in `_set_ocr()`.
So it seems we need a little more sophisticated solution here, although I have no idea how this would look like.

DerAndere1, 2025-03-30:
I updated the pr.if you have time you can Test it. I understand you have to backport it again, sorry for the oncomvenience. My own Test system is still in my old flat, so it will Take some time until i can Test my new solution myself.

closed 2025-04-14 by thinkyhead