MarlinFirmware/Marlin #27728: [BUG] LCD_LANGUAGE does not work anymore

Amachete, 2025-03-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello

I am updating with latest version of Marlin and the translations on the LCD does not work anymore, I have tried several revisions. texts are only in english.

Bugfix (latest):  Does not work
release-2.1.3-beta2: Does not work
release-2.1.3-beta1: Does not work

Marlin-2.1.2.5: WORKS

THe only thing I change to build and upload to see it this works

#define REPRAP_DISCOUNT_SMART_CONTROLLER

#define LCD_LANGUAGE es//en

If you need something else let me know.

### Expected behavior

LCD texts translated to my language es (spanish)

### Actual behavior

LCD texts only in English

### Version of Marlin Firmware

latests

### Printer model

p3steel

### Electronics

ramps

### LCD/Controller

reprapcontroller

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_BUGFIX_P3Steel.zip](https://github.com/user-attachments/files/19142446/Marlin_BUGFIX_P3Steel.zip)

ellensp, 2025-03-08:
In Marlin/src/inc/Conditionals-2-LCD.h  is

```cpp
#if NONE(HAS_MARLINUI_U8GLIB, HAS_TFT_LVGL_UI, TFT_COLOR_UI, DGUS_LCD_UI_E3S1PRO)
  #undef LCD_LANGUAGE
#endif
```

this is removing LCD_LANGUAGE for your display

As a quick fix remove those 3 lines

ellensp, 2025-03-08:
A PR to fix this has been created

closed 2025-03-08 by ellensp