MarlinFirmware/Marlin #27732: [BUG] MKS_ROBIN_NANO_V3 SD-Card and USB drive are not detected

anomalchik, 2025-03-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After updating the firmware from the bugfix-2.1.x branch from 2024-10-21 to 2025-03-07, the drives stopped being detected. It also causes random reboots.

### Bug Timeline

2024-10-21 - Work, 2025-03-07 - Doesn't works

### Expected behavior

Detecting SD or USB drive when it is connected

### Actual behavior

Not detecting. It also causes random reboots during connect a USB flash drive.

### Steps to Reproduce

1. #define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
2. #define SDSUPPORT
3. #define USB_FLASH_DRIVE_SUPPORT
4. #define USE_OTG_USB_HOST
5. #define MULTI_VOLUME
6. Build mks_robin_nano_v3_usb_flash_drive_msc variant.

### Version of Marlin Firmware

buigfix-2.1.x 2025-03-07

### Printer model

Custom

### Electronics

MKS_ROBIN_NANO_V3

### LCD/Controller

MKS_TS35_V2_0

### Other add-ons

none

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/user-attachments/files/19157935/config.zip)

thinkyhead, 2025-04-20:
I've got this board so I'll do some testing to see if I can reproduce the issue, then go from there.

thinkyhead, 2025-04-21:
I also have the TS35 so I was able to do a pretty close test. Based on Keith's input I updated the `bugfix-2.1.x` conditionals. I also updated your configs for the upcoming 2.1.3-b3 so here they are:

[configs-updated.tar.gz](https://github.com/user-attachments/files/19827741/configs-updated.tar.gz)

If you use these to do a new build with the latest code the file navigation now seems to work pretty well (except the cosmetic bug where portions of the screen are not being cleared).

I will see if I can fix those cosmetic issues, and in the meantime give the updated firmware a test and see if it works for you.

closed 2025-05-22 by thinkyhead