MarlinFirmware/Marlin #27743: [FR] Native GD32 support for Creality 4.2.2 Boards with this MCU

bmourit, 2025-03-13:
### Is your feature request related to a problem? Please describe.

Nt really a huge problem, but it there are differences between GD32 and STM32.

### Are you looking for hardware support?

Yes

### Describe the feature you want

I am interested in upstreaming my native GD32F303RET6 support that I have been using for a while now. Firstly, this does not use any Gigadevice provided libraries, but one I wrote from scratch along with some CMSIS Headers. This started a little bit like an experiment, but mainly from frustration with getting a working Marlin port using the GD32CommunityCore Arduino core that uses the GD32 SPL. There are a bunch of issues with the SPL, and the core tried to work around these, but after so long I decided it would be easier to start from scratch, so I did. All peripherals from this MCU are supported the the MFL (Microcontroller Firmware library) except for CAN, which I will be adding eventually, but this board doesn't use CAN anyway.

I made a new Arduino Core for this that uses the custom MFL library and since the end goal was with Marlin in mind, I was able to develop them in tandem with the Marlin HAL portions. The Arduino core also follows the latest Arduino Core API.

All are LGPLv3 to make things simple, including the library, making it easy to add new low level functionality when needed/wanted.

Since I only have an Ender 3 Board V4.2.2 with the GD32F303RET6 MCU, I have only been able to test this printer. However, I believe all printers using this board will have the exact same pinout, meaning there should be only minor work needed to use this in other printers as well.

Is this something the community would be interested in?

While the STM32 libraries and drivers work alright with GD32 chips, there are some differences, and therefor benefits to using native support.

- CPU core speed increase (120 MHz)
- Faster GPIO
- The STM32 library sets bits that GD32 does not have (Eg. ADC resolution changes with STM32 does nothing - so it is fixed at 12Bit)
- Proper startup voltage and stabilization bits are set in hardware.
- Configurable Hardware Oversampling for ADC (default is 16x)
- FPU offloading

In general, performance is improved and some extra features are available.

Debugging is also a lot easier (for low level) since I wrote the MFL library in C++ (This was the experiment part).
The binary size for the latest marlin was nearly identical with STM32 (MFL is some bytes less) and RAM usage is a little less I believe (I need to verify this).

I am happy to make a pull request, and I have been preparing for possible upstreaming for a little while now. Before I go any farther though I thought it would be a good idea to see if this would even be welcome.

Same Marlin Feature set should be supported as well. For example Endstop interrupt feature, SDCard DMA, and UART with DMA on rx. In fact, the latter two require DMA.

thisiskeithb, 2025-03-13:
> I am interested in upstreaming my native GD32F303RET6 support that I have been using for a while now.

Please open a PR for review 🙂

> Is this something the community would be interested in?

I’m sure many users will appreciate native GigaDevice support.

bmourit, 2025-03-14:
Will do! Thanks for the feedback.

closed 2025-03-28 by thinkyhead