MarlinFirmware/Marlin #27746: Z2 stepper motor is only in sync with Z stepper motor when using 16 microsteps

Nico9n, 2025-03-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am using 2 stepper motors and drivers (TMC2226 in UART mode) for the Z axis. When I use #define Z_MICROSTEPS 16, everything works as expected. However, when I use #define Z_MICROSTEPS 32 or any other value higher than 16, Z2 spins too fast. Z2 is not in sync with Z.

### Expected behavior

Z2 should spin at the same speed as Z.

### Actual behavior

Z2 spins faster than Z.

### Steps to Reproduce

1. Use a printer with separate drivers and motors for Z and Z2.
2. Use TMC drivers in UART mode.
3. Set #define Z_MICROSTEPS 32 or any value higher than 16 in Configuration_adv.h.
4. Adjust the values for DEFAULT_AXIS_STEPS_PER_UNIT depending on the microsteps in Configuration.h 
5. Home the Z axis. You will see that Z2 spins too fast.

### Version of Marlin Firmware

2.1.x 

### Printer model

Prusa i3 clone with changes

### Electronics

Manta m8p v2 Mainboard

### LCD/Controller

Mini 12864 v2.0

### Bed Leveling

None

### Your Slicer

Simplify3D

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/user-attachments/files/19272601/config.zip)

[m122output.txt](https://github.com/user-attachments/files/19272605/m122output.txt)

robherc, 2025-03-16:
Maybe you nred to set the microsteps on Z2 to match the new microsteps on Z?

Nico9n, 2025-03-16:
I tried your suggestion, but the result is the same.

in Configuration_adv.h.
`#define Z_MICROSTEPS 32
#define Z2_MICROSTEPS 32`

I think the debug output from M122 indicates that the steps are set correctly:
`X   Y   Z   Z2  
msteps   32  32  32  32`

ellensp, 2025-03-16:
This is literally not possible. The code is such that for every step on Z a step is also taken on Z2.

Most probable cause is you have a dead stepper driver that is not accepting changed microstepping.

Swap X and Y stepper driver with Z and Z2, and see if the issue on Z moves to X or Y

ellensp, 2025-03-16:
I do see a bug,  sadly I don't have this hardware to try it on.

The issue I see is that Marlin/src/pins/stm32h7/pins_BTT_MANTA_M8P_V2_0.h has  #define Z2_SERIAL_TX_PIN                  PB5
So it expects Z2 uart to always be on M4 even if your not using M4 for Z2

Ie it has 
```
  #define X_SERIAL_TX_PIN                   PC13      // M1
  #define Y_SERIAL_TX_PIN                   PE3       // M2
  #define Z_SERIAL_TX_PIN                   PB9       // M3
  #define Z2_SERIAL_TX_PIN                  PB5       // M4
  #define E0_SERIAL_TX_PIN                  PG14      // M5
  #define E1_SERIAL_TX_PIN                  PG10      // M6
  #define E2_SERIAL_TX_PIN                  PD5       // M7
  #define E3_SERIAL_TX_PIN                  PC6       // M8

``` 

But this breaks the uart pins is your not using the stepper drivers like that

It should be

```
  #define X_SERIAL_TX_PIN                   PC13      // M1
  #define Y_SERIAL_TX_PIN                   PE3       // M2
  #define Z_SERIAL_TX_PIN                   PB9       // M3
  #define E0_SERIAL_TX_PIN                  PB5       // M4
  #define E1_SERIAL_TX_PIN                  PG14      // M5
  #define E2_SERIAL_TX_PIN                  PG10      // M6
  #define E3_SERIAL_TX_PIN                  PD5       // M7
  #define E4_SERIAL_TX_PIN                  PC6       // M8
``` 
And then Z2 is auto set to correct E0 serial pin 

Update your Marlin/src/pins/stm32h7/pins_BTT_MANTA_M8P_V2_0.h with these changes and see if it fixes your issue

thisiskeithb, 2025-03-16:
This is another one of the boards that follows the original implementation set by the Fysetc Spider board of defining a Z2 instead of everything past Z/Z1 being an extruder driver.

ellensp, 2025-03-16:
No it doesn't. there are no other Z2 pins being defined in pins_BTT_MANTA_M8P_V2_0

thisiskeithb, 2025-03-16:
> No it doesn't. there are no other Z2 pins being defined in pins_BTT_MANTA_M8P_V2_0

There's a Z2 pin literally defined in the pins file 😄 Direct link to pins file / current commit is below: 

https://github.com/MarlinFirmware/Marlin/blob/b954959f55d86e0feff045a561f70524bb260bc8/Marlin/src/pins/stm32h7/pins_BTT_MANTA_M8P_V2_0.h#L320

ellensp, 2025-03-16:
Yes that is the issue.
But that only works if the Z2_STEP_PIN, Z2_DIR_PIN and Z2_ENABLE_PIN are also hardcoded into pins_BTT_MANTA_M8P_V2_0
They are not present.
So this board does not have a statically defined Z2!

ellensp, 2025-03-16:
user also has 
//#define E0_DRIVER_TYPE TMC2209
#define E1_DRIVER_TYPE TMC2209
Which is not valid, you can't have a E1 without a E0

Nico9n, 2025-03-17:
> I do see a bug, sadly I don't have this hardware to try it on.
> 
> The issue I see is that Marlin/src/pins/stm32h7/pins_BTT_MANTA_M8P_V2_0.h has #define Z2_SERIAL_TX_PIN PB5 So it expects Z2 uart to always be on M4 even if your not using M4 for Z2 This would be fine if all z2 pins was set to M4 permanently but it is not.
> 
> Ie it has
> 
> ```
>   #define X_SERIAL_TX_PIN                   PC13      // M1
>   #define Y_SERIAL_TX_PIN                   PE3       // M2
>   #define Z_SERIAL_TX_PIN                   PB9       // M3
>   #define Z2_SERIAL_TX_PIN                  PB5       // M4
>   #define E0_SERIAL_TX_PIN                  PG14      // M5
>   #define E1_SERIAL_TX_PIN                  PG10      // M6
>   #define E2_SERIAL_TX_PIN                  PD5       // M7
>   #define E3_SERIAL_TX_PIN                  PC6       // M8
> ```
> 
> But this breaks the uart pins is your not using the stepper drivers like that
> 
> It should be
> 
> ```
>   #define X_SERIAL_TX_PIN                   PC13      // M1
>   #define Y_SERIAL_TX_PIN                   PE3       // M2
>   #define Z_SERIAL_TX_PIN                   PB9       // M3
>   #define E0_SERIAL_TX_PIN                  PB5       // M4
>   #define E1_SERIAL_TX_PIN                  PG14      // M5
>   #define E2_SERIAL_TX_PIN                  PG10      // M6
>   #define E3_SERIAL_TX_PIN                  PD5       // M7
>   #define E4_SERIAL_TX_PIN                  PC6       // M8
> ```
> 
> And then Z2 is auto set to correct E0 serial pin
> 
> Update your Marlin/src/pins/stm32h7/pins_BTT_MANTA_M8P_V2_0.h with these changes and see if it fixes your issue

I have updated the pins_BTT_MANTA_M8P_V2_0.h file, and it fixes my issue. Thank you! :)

closed 2025-03-18 by ellensp