MarlinFirmware/Marlin #27749: [BUG] (bug summary) ender 3 max board selection. BTT SKR mini e3v3

snoboardin77, 2025-03-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

can't compile a bin for ender 3 max using a bit skr mini e3v3 board. tried to modify a ender 3 pro but still has a bunch of issues and ender 3 won't let me change the board to btt skr mini e3v3

### Bug Timeline

ive been dealing with for 2 months and Facebook group can't help 

### Expected behavior

I expected to have a working printer 

### Version of Marlin Firmware

bugfix2.1.x

### Printer model

ender 3 max

### Electronics

BTT SKR MINI E3V3 

### LCD/Controller

I have the stocker ender 3 max display but would like to use the BTT TFT35

### Other add-ons

CR touch 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2025-03-17:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server

closed 2025-03-17 by thisiskeithb

ellensp, 2025-03-17:
And he came discord where he was getting help...

ellensp, 2025-03-17:
@snoboardin77 come back to the discord. I have created configs for you to try.