MarlinFirmware/Marlin #27756: [BUG] (Compatibility 2 types of screens)

BravusManCo, 2025-03-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, 
I'm updating the firmware of my machine to the new marlin version.
The main problem is that y have the SKR V 1.4 Turbo with the LCD touchscreen TFT35 and compiling with visual studio code. When compiling it obligates me to assing a type of classic LCD and a TFT (since it has  the both options, but I just use the touch one because a roll of fillament fell on top and the roullete doesnt work anymore)
The real question is that when I'll finish compiling (still trying to figure out) I want to switch the thermistor probes from E0 to E1 because is in shortcircuit.

### Bug Timeline

Today, when cleaning the nozzle  guess short circuit of thermistor

### Expected behavior

I would like to compile with just the touchscreen and the pin changed from pins_BTT_SKR_V1_4.h but cant choose both options of screen or know what pin to change.

### Actual behavior

Compiling to update it and use the machine without the error that kills or halted my printer

### Version of Marlin Firmware

Marlin-2.1.2.5

### Printer model

Custom AM8 blv

### Electronics

SKR V1.4 Turbo

### LCD/Controller

TFT35

### Other add-ons

Octoprint

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-2.1.2.5.zip](https://github.com/user-attachments/files/19406160/Marlin-2.1.2.5.zip)

thisiskeithb, 2025-03-22:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server

closed 2025-03-22 by thisiskeithb

ellensp, 2025-03-22:
Hint: the btt tft is not a marlin controlled tft. It is a serial host with a tft screen.  It just needs a serial port. + the enabling the 128x64 like normal.

BravusManCo, 2025-05-19:
Great! Finnaly up and printing again. Just used the tip from @ellensp where I put the serial number -1 for the emulated usb to connect via octopi and then the 0 to enable the tft to send and receive messages.
Thanks!