MarlinFirmware/Marlin #27757: Part Cooling Fan for CTC Bizer

MechEng2016, 2025-03-25:
### Is your feature request related to a problem? Please describe.

I have installed the Marlin 2.1.2.5 for the CTC Bizer but found the Part-Cooling Fan won't activate. I have looked through the code and don't find the feature listed there. The fan plugs into the "EXTRA" slot of the MightyBoard RevE. Maybe I haven't seen it. Any help would be appreciated.

### Are you looking for hardware support?

CTC Bizer printers don't come with a part cooling fan. It is added on as an extra. The MightyBoard RevE uses the "EXTRA" slot to support the fan.

[image]

### Describe the feature you want

I hoping to get some code to activate the part-cooling fan that plugs in the "Extra" slot.

[image]

thisiskeithb, 2025-03-25:
The code is already there, but it may not work on *your* board (see comment): https://github.com/MarlinFirmware/Marlin/blob/14fa7057718acbed8d0e4238f2eba9b9f645db64/Marlin/src/pins/mega/pins_MIGHTYBOARD_REVE.h#L156

closed 2025-03-25 by thisiskeithb

MechEng2016, 2025-03-27:
The fan worked before I put on the Marlin code. Is there any other code
I need to activate?

ellensp, 2025-03-27:
Part cooling fan is always on FAN0_PIN

You need to tell it to use MOSFET_F_PIN
Ie Add #define FAN0_PIN MOSFET_F_PIN  to your Config
(in older  marlins its FAN_PIN)