MarlinFirmware/Marlin #27758: [BUG] (bug summary) Dual Z stops Z1 from working

TaiTair, 2025-03-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Config files: 

[Configuration.zip](https://github.com/user-attachments/files/19440779/Configuration.zip)

This is with an SKR PRO 1.2 on 2.1.5 bugfix from the official site.

Enabled Dual Z by setting Z2_DRIVERTYPE to TMC2208. All my drivers are TMC2208. Then I plugged my second Z motor into E1.
Now only my second Z motor spins. The first one is entirely useless. At first I thought there was a problem with the pins so I modified my pins file to set the E1 pins to Z2. Still the same thing.

I tried the options from the documentation like NUM_Z_STEPPERS or DUAL_Z_STEPPERS but that seems to be deprecated now. Couldn't even compile my firmware. It seems like I'm missing something.

Here's the interesting thing though. If I plug my second motor into my Z1 port (Just in parallel of the Z0 on the motherboard) then both the motors work. But now they are in parallel which makes g34 impossible. It seems like as soon as something gets plugged into E1 then Z becomes useless.

### Bug Timeline

I'm not sure. I've only tried using G34 now.

### Expected behavior

Expected both my Z motors to spin after enabling dual Z.

### Actual behavior

Only the motor on E1 spins

### Steps to Reproduce

Use my configuration on a skr pro 1.2 with tmc2208s I guess and see if it works. Adjust your bed size first cause my printer is a decent size.

### Version of Marlin Firmware

2.1.5 bugfix from the official site

### Printer model

Heavily modified FLSUN C5 (not produced anymore)/Basically a custom CR-10

### Electronics

SKR PRO 1.2 with TFT35-E3 V3.0. FLSUN motors, Sprite Pro extruder kit and Creality heated bed.

### LCD/Controller

TFT35-E3 V3.0

### Other add-ons

BLTOUCH

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2025-03-25:
Make sure you've added jumpers to the second Z port tied to the first Z driver since it's an onboard splitter connected in series.

[image]

From: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/SKR-PRO-V1.2/manual/SKR-PRO-V1.2-user%20manual.pdf

closed 2025-03-25 by thisiskeithb

TaiTair, 2025-03-25:
Thank you so much.

thinkyhead, 2025-03-27:
@thisiskeithb — BTT certainly documents this, but perhaps we should mention this Z axis quirk in the pins files too as a reminder to ourselves.