MarlinFirmware/Marlin #27761: [BUG] PIDTEMPBED heats to 10° below target, then panics

FlyingRatBull, 2025-03-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using `PIDTEMPBED` (with tuning through `M303 E-1 C8 S60`) the bed doesn't heat to the right temperature and the printer then panics because of stuck temp sensor:

1. `M140 S60`
2. Bed is heated to 50°C and stays there consistently
3. After a while the printer panics with _Heating Bed failed_ because of an apparent stuck Thermistor (because the bed temp doesn't rise I guess)

* Bang-Bang works just fine
* Happens with OctoPrint and through the LCD menu
* Manual temperature measurement of the bed confirms ~50°C

### Bug Timeline

At least since 2.1.2.5

### Expected behavior

* Set target temp to 60°
* Bed heats to 60° without panicking

### Actual behavior

* Bed heats to 50° then panics

### Steps to Reproduce

1. Enable `PIDTEMPBED`
2. Tune with `M303 E-1 C8 S60`
3. `M140 S60`

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Mar 27 2025 11:45:26)

### Printer model

Anycubic i3+

### Electronics

Stock bed with magnetic metal plate on top and wool/aluminum insulation on the bottom

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/user-attachments/files/19485008/Config.zip)

ellensp, 2025-03-27:
This is a limitation of your hardware

You need to adjust the following to fit your hardware so it doesn't trip 
```Cpp
**
 * Thermal Protection parameters for the bed are just as above for hotends.
 */
#if TEMP_SENSOR_BED && ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD        20 // (seconds)
  #define THERMAL_PROTECTION_BED_HYSTERESIS     2 // (°C)

  /**
   * As described above, except for the bed (M140/M190/M303).
   */
  #define WATCH_BED_TEMP_PERIOD                60 // (seconds)
  #define WATCH_BED_TEMP_INCREASE               2 // (°C)
#endif
```

The most beneficial fix it to insulate under your heated be, so that your not just trying to heat the room.

ellensp, 2025-03-27:
I do not think this is a bug.  but user is instant.

ellensp, 2025-03-27:
I tested these configs as much as possible on the SIM,  If left for a while it does error but

[image]

But this is from   #define THERMAL_PROTECTION_VARIANCE_MONITOR

From the Configuration_adv.h file

```
  * Thermal Protection Variance Monitor - EXPERIMENTAL
   * Kill the machine on a stuck temperature sensor.
   *
   * This feature may cause some thermally-stable systems to halt. Be sure to test it thoroughly under
   * a variety of conditions. Disable if you get false positives.
```

FlyingRatBull, 2025-03-27:
I just tested it without `THERMAL_PROTECTION_VARIANCE_MONITOR` and the bug is still present.

As you can see, it seems like there would be enough power, but the heater gets stopped at exactly 50°:

[image]

ellensp, 2025-03-27:
I cannot replicate that error.

I suspect your simply not actually updating the PID values

You said "Tune with M303 E-1 C8 S60" 
But that only shows what the pid should be, it  does not update or use it without U flag.   (see https://marlinfw.org/docs/gcode/M303.html)
When you then "M140 S60" it uses old pid values from eeprom

Try  M303 E-1 C8 S60 U1
then M140 S60

FlyingRatBull, 2025-03-27:
That didn't work either and anyway I also reset the EEPROM before the last PID tuning/config setting:

```
M303 E-1 C8 S60 U1
PID Autotune start
 bias: 161 d: 93 min: 58.75 max: 60.28 T:20.67 /0.00 B:59.30 /0.00 @:0 B@:127
 bias: 161 d: 93 min: 59.15 max: 60.36 T:20.73 /0.00 B:59.92 /0.00 @:0 B@:127
 bias: 123 d: 123 min: 59.90 max: 60.83 Ku: 334.10 Tu: 17.20
 No overshoot
 Kp: 66.82 Ki: 7.77 Kd: 383.17
 bias: 96 d: 96 min: 59.90 max: 60.94 Ku: 234.68 Tu: 13.11
 No overshoot
 Kp: 46.94 Ki: 7.16 Kd: 205.07
 bias: 94 d: 94 min: 59.79 max: 60.39 Ku: 397.72 Tu: 10.49
 No overshoot
 Kp: 79.54 Ki: 15.17 Kd: 278.03
 bias: 86 d: 86 min: 59.84 max: 60.56 Ku: 307.66 Tu: 11.14
 No overshoot
 Kp: 61.53 Ki: 11.05 Kd: 228.51
 bias: 79 d: 79 min: 59.84 max: 60.42 Ku: 351.14 Tu: 11.14
 No overshoot
 Kp: 70.23 Ki: 12.61 Kd: 260.80
 bias: 75 d: 75 min: 59.84 max: 60.36 Ku: 370.81 Tu: 10.81
 No overshoot
 Kp: 74.16 Ki: 13.72 Kd: 267.33
PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
#define DEFAULT_bedKp 74.16
#define DEFAULT_bedKi 13.72
#define DEFAULT_bedKd 267.33
ok

M140 S60
ok
Error:Heating failed, system stopped! Heater_ID: bed
```

FlyingRatBull, 2025-03-27:
I just tested `M140 S50` and `M140 S55` and it stayed at 40°/45° respectively. So it seems to be a constant 10° offset.

FlyingRatBull, 2025-03-27:
> 10C is the PID functional range. That means the PID tuning is bad. If you cannot get a successful PID auto tune on the bed, double the P parameter and try again.

Do I have to reset my EEPROM for this, or is defining in config/updating firmware enough?

InsanityAutomation, 2025-03-27:
You're fine setting this by G-Code or on the menus, and then store. Anyway to get the value into that actual setting. You don't necessarily need to reset the whole eeProm unless you are changing the value in the config file and reflashing so it's now the default, but that's a lot more effort than just updating it by g-code or menu.

FlyingRatBull, 2025-03-27:
Ok, I think I found the problem:

I just did `M304 P77.86 I14.40 D280.67` (so the original P value) and the heating worked as expected.

I previously __only__ used the configuration values/rebuild/reset EEPROM after the autotune step - without using `M304` as that's what's stated in the autotune epilogue. This breaks the PID for me. Is this not the way that's intended? Do I __have to__ use M304 regardless of config value?

```
M304
  M304 P0.00 I0.00 D0.00
ok

M304 P77.86 I14.40 D280.67
ok

M304
  M304 P77.86 I14.40 D280.67
ok

M140 S55
ok

M140 S60
ok
```

closed 2025-03-27 by thisiskeithb

FlyingRatBull, 2025-03-28:
Thanks for closing the bug already...again. As there is still one question remaining and a potential bug:

* Do I have to use M304 regardless of config value?
* `M303 E-1 C8 S60 U1` __didn't work__ even though it ought to be setting the appropriate values for M304