MarlinFirmware/Marlin #27763: [BUG] Blobs (ovextrusion?)

vovodroid, 2025-03-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Latest bugfix-2.1.x version (EDITABLE_HOMING_CURRENT) causes blob formation (looks like overextrusion). Version Bump distribution date (2025-03-07) works fine.

The same gcode file:
[image]

[image]

I use firmware retraction, and tried to increase it up to 2mm during print with no effect, but extruder started to produce loud clicking sound on retract/unretract (no such sound in old vesrion with the same retract).

### Bug Timeline

After [cron] Bump distribution date (2025-03-07) commit

### Version of Marlin Firmware

Commit 460bf112d8a52c67055c253569023178af8dd218

### Electronics

SKR-3, TMC2209

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/19501912/Marlin.zip)

vovodroid, 2025-03-28:
Could it be changes in ABL or TMC flags commits?

ellensp, 2025-03-29:
perhaps related to https://github.com/MarlinFirmware/Marlin/issues/27769  no retraction would cause blobs

ellensp, 2025-03-29:
closing this issue  as https://github.com/MarlinFirmware/Marlin/issues/27769 is a better description and a fix.

closed 2025-03-29 by ellensp