MarlinFirmware/Marlin #27766: [FR] (Marlin-2.1.2.5 for printrbot or printrboard rev. F4)

mightymos, 2025-03-28:
### Is your feature request related to a problem? Please describe.

Apparently a github gist does not allow file attachments so...
I just wanted a compiled Marlin HEX for printrbot/printrboard to be available somewhere on the web.
I do not even know if my used printer works yet, but HEX flashes successfully and serial outputs version information.

Apparently there is no pre-compiled build for this printer model:
https://github.com/MarlinFirmware/MarlinBuilds/tree/release-2.1.2.5/builds

However I merely used the configuration file which is available.

Upload hex to Atmel 90USB1286 in DFU mode using these instructions and Atmel FLIP tool:

https://reprap.org/wiki/Printrboard
https://ww1.microchip.com/downloads/en/Appnotes/doc7769.pdf
https://www.microchip.com/en-us/development-tool/FLIP

Serial port output after flashing:

```
start
External Reset
Marlin 2.1.2.5
echo: Last Updated: 2024-11-18 | echo:No SD card                              
```

[Marlin_2.1.2.5_printrbot_revf4.ino.zip](https://github.com/user-attachments/files/19511075/Marlin_2.1.2.5_printrbot_revf4.ino.zip)

### Are you looking for hardware support?

Printrbot or Printrboard rev. F4

### Describe the feature you want

It would be cool if a precompiled hex could be produced.
There must be some reason/history why it is not despite the presence of a configuration file for this model printer.

thisiskeithb, 2025-03-28:
You should open an issue in the MarlinBuilds repo, not here.

https://github.com/MarlinFirmware/MarlinBuilds/issues

closed 2025-03-28 by thisiskeithb

mightymos, 2025-03-28:
https://github.com/MarlinFirmware/MarlinBuilds/issues/3