MarlinFirmware/Marlin #27767: [BUG] Building with GCC14 and C++23 causes 'volatile-qualified left operand' warnings

bmourit, 2025-03-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a patch to fix the warnings prepared as well, so this bug report is just to track that. Fairly simple, just two places in the planner.h.cpp files.

### Bug Timeline

Introduced when building with newer GCC and C++ standards

### Expected behavior

No warnings, and remove the need for using -Wno-volatile for the GD32 platform.

### Actual behavior

Two instances of the warning are generated:
      "warning: using value of assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]"

### Steps to Reproduce

Remove the flag -Wno-volatile from the gd32.ini file and build a GD32 board.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender-3

### Electronics

v4.2.2 MFL

### LCD/Controller

Default

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2025-03-29:
> I have a patch to fix the warnings prepared as well, so this bug report is just to track that.

To save you some time, you can skip creating a bug report if you already have a patch / PR ready to go.

bmourit, 2025-03-29:
Excellent!!! I'm glad to hear that! Thanks!

closed 2025-04-01 by thinkyhead