MarlinFirmware/Marlin #27769: [BUG] Missing E[x]_DIR_INIT() in stepper.cpp after commit add72fd

rondlh, 2025-03-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Extruder only turns in one direction after commit add72fd.
From the Commit diffs I can see that  E0_DIR_INIT(),  E1_DIR_INIT() etc. were lost after commit add72fd

Restoring these functions solves my issue.

### Bug Timeline

#add72fd

### Expected behavior

Extruder can turn in both directions.

### Actual behavior

Extruder only turns in one direction.

### Steps to Reproduce

Just try to move the extruder in both directions

### Version of Marlin Firmware

Latest bugfix (March 28 2025)

### Printer model

Custom

### Electronics

Fly D8 Pro (STM32H723)

### LCD/Controller

BTT TFT35

### Other add-ons

CAN toolhead

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

These are just the default configs...
[Configuration.zip](https://github.com/user-attachments/files/19517110/Configuration.zip)

vovodroid, 2025-03-29:
Does it look like #27763?

[image]

[image]

vovodroid, 2025-03-29:
> E0_DIR_INIT(), E1_DIR_INIT() etc. were lost after commit [add72fd](https://github.com/MarlinFirmware/Marlin/commit/add72fd0452addb13ac7c1fea7d7476dd991cd93)

Does it mean that retraction actually works as unretraction?

vovodroid, 2025-03-29:
It also has double `TERN_(HAS_K_DIR,  K_DIR_INIT());`

ellensp, 2025-03-29:
@vovodroid 
can you try https://github.com/MarlinFirmware/Marlin/pull/2777  It looks good in my head and it builds...

rondlh, 2025-03-29:
> Does it look like [#27763](https://github.com/MarlinFirmware/Marlin/issues/27763)?
> 
> [image]
> 
> [image]

I don't know, in my case it always retracts.

ellensp, 2025-03-30:
fix merged

closed 2025-03-30 by ellensp