MarlinFirmware/Marlin #27771: [BUG] High resend ratio

FlyingRatBull, 2025-03-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After switching from 2.1.2.5 to bugfix-2.1.x

### Expected behavior

GCode commands get sent to the printer without needing to be resent.

### Actual behavior

Resend ratio immediately spikes to 10-20% and commands need to get resent all throughout the print.

### Steps to Reproduce

1. Switch to bugfix-2.1.x
2. Print anything

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Anycubic-i3+

### Electronics

Stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

* On 2.1.2.5: Not a single resend for 100s of prints
* On bugfix-2.1.x: Spikes to 10-20% at the start of the print, levels out at ~5% during the print (with periodic resends throughout the print)
* I already ruled out:
  * Defective cable
  * Defective Host PC
  * Defective Host Software
  * Defective Printer Hardware
  * Too high baud rate
  * After downgrading to 2.1.2.5 the resends are completely gone

[Config.zip](https://github.com/user-attachments/files/19526737/Config.zip)

FlyingRatBull, 2025-05-17:
No one? Anyone?

RenateUSB, 2025-07-08:
I think one resend is one too many. I bought a used Sovol SV01 BTT SVR Mini 1.2 with 2.0.9.1. The USB serial was balky often getting "k" for "ok" so I upgraded to a new build of bugfix-2.1.x. If I run it without line numbers it's ok, but with line numbers it hits resends sometimes. I wonder how many people run without line numbers and don't notice that one G1 out of 50,000 went missing.

ellensp, 2025-07-08:
"one resend is one too many" is unrealistic

"If I run it without line numbers it's ok"  without line number there are no checksums, so marlin cannot detect errors.

RenateUSB, 2025-07-09:
Still getting "k" for "ok" occasionally.
USB hardware protocol analyzer shows messages sent from Marlin chopped up in nonsensical ways.
Quite often the first packet is just "o" from "ok".
It's neither efficient or logical.
It would be nice if USB serial responses were canonical.

[image]

Also `N123 M105` does not get a numbered ok.

It's not clear if @FlyingRatBull is using USB serial or wired serial.

FlyingRatBull, 2025-07-09:
> Still getting "k" for "ok" occasionally. USB hardware protocol analyzer shows messages sent from Marlin chopped up in nonsensical ways. Quite often the first packet is just "o" from "ok". It's neither efficient or logical. It would be nice if USB serial responses were canonical.
> <img alt="Image" width="511" height="112" src="https://private-user-images.githubusercontent.com/35894241/464336330-073fdf22-bb39-42d9-a9a0-3c3ff48051e1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIwODcxODAsIm5iZiI6MTc1MjA4Njg4MCwicGF0aCI6Ii8zNTg5NDI0MS80NjQzMzYzMzAtMDczZmRmMjItYmIzOS00MmQ5LWE5YTAtM2MzZmY0ODA1MWUxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzA5VDE4NDgwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFlZjk0NTZiYzhiZGQ0OGE0N2RhOTZlNGE1MTBkM2IxYjcwY2JkNmM4ZmU5Yzg4OGM0NmVhNjJjYTkwMzdkZjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.QwUFsj7zqOA1j1zB2UGpj7SQsw3ZR1r1WyoqEo-zmZw">
> 
> Also `N123 M105` does not get a numbered ok.
> 
> It's not clear if [@FlyingRatBull](https://github.com/FlyingRatBull) is using USB serial or wired serial.

I'm using USB serial with the default baud rate and the exact same configuration which worked before with 2.1.2.5.

RenateUSB, 2025-07-09:
"Baud rate" has no real meaning when talking about a device that is only pretending to be a UART.

RenateUSB, 2025-07-10:
Urps, you're right. They're only running a 2560.

ellensp, 2025-07-10:
"#define TX_BUFFER_SIZE 0" Why is this 0 ?   this means TX is not buffered at all "#define TX_BUFFER_SIZE 0"

Try upping it to 4

FlyingRatBull, 2025-07-10:
> "#define TX_BUFFER_SIZE 0" Why is this 0 ? this means TX is not buffered at all "#define TX_BUFFER_SIZE 0"
> 
> Try upping it to 4

That's because it's [0 upstream](https://github.com/MarlinFirmware/Configurations/blob/a7d52ed1c90e3741296c4b26954eaba2702e976c/config/default/Configuration_adv.h#L2692).

Nonetheless if it was working with 2.1.2.5 with `TX_BUFFER_SIZE 0`, changing this would just circumvent the bug, not fix it.

RenateUSB, 2025-07-11:
> regarding 2.

Does that mean you **somewhat** agree on some of the others? :p

> Have you never heard of electrical interference, or cheap cables?

I presume that most people who aren't using processor USB are using USB UARTs where the TX and RX lines are running only inches.
If you are actually running RS-232 many meters it is best to use decent cables.

I swore that the big print that I just started would run perfectly without a glitch.
But it bombed. On a network glitch between Windows and RPi. Oops.
6) Don't run a big job without:
```
nohup stdbuf -o0 -e0 ./squid vase.gcode >> vase.log 2>&1
```
I'm trying again.
```
Bed heater set to 60
  0%
Received unnumbered ok for line 2
Bed reached 60
Hotend heater set to 200
Received unnumbered ok for line 5
Hotend reached 200
Homed
Fan turned off
  1%
Fan set to 100%
  2%
  3%
  4%
  5%
  6%
  7%
```

RenateUSB, 2025-07-17:
A week gone by. I've printed a bunch of stuff. Not a single "resend" (yet).
Ok, it's never possible to prove that you fixed an intermittent, but I'm happy.
My server is designed to explode if anything asks for a resend, so I would have noticed it.

AFAIK, the actual "serial" buffering for USB CDC interface is unrelated/not affected by anything in Conf~.h or Conf~_adv.h.
It is all hardwired into the code supporting the USB CDC for the processor.
Since a STM32F103RCT6 has 48 kbytes RAM, it seems foolish to be too parsimonious.
You're just leaving RAM on the table.

Sorry, this doesn't have much to do with the OP. But it should give you hope that life without resend is possible.