MarlinFirmware/Marlin #27775: [BUG] GD32 MFL cannot build without U8glib update

bmourit, 2025-03-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The initial support for GD32 that was merged had the pull request in the U8glib repo linked in, so I'm not sure if it was intentionally merged anyway without it - perhaps pending LCD changes? If so, this can be closed. If it was just accidentally missed, then that repo has the feature request and patch already requested. I'm only creating the bug report here since Marlin wont build without the changes.

### Expected behavior

Successful build.

### Actual behavior

Errors with build due to the ARDUINO define and the library not excluding ARDUINO_ARCH_MFL in various files.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender-3

### Electronics

GD32

### LCD/Controller

Stock

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

The U8glib patch hereL https://github.com/MarlinFirmware/U8glib-HAL/pull/35

bmourit, 2025-04-02:
Progress! I see this was merged, so now we'll just wait for the new package to be released. Awesome! And thank you!

ellensp, 2025-04-12:
If you change that to the following it uses latest code vs latest release 
HAS_MARLINUI_U8GLIB = https://github.com/MarlinFirmware/U8glib-HAL/archive/refs/heads/master.zip

Lazar0v, 2025-04-18:
This is a log when compiling with modified Configuration.h , Configuration_adv.h
With base files it compiles without problems
...............
Compiling .pio\build\GD32F303RE_creality_mfl\src\src\gcode\bedlevel\G26.cpp.o
Compiling .pio\build\GD32F303RE_creality_mfl\src\src\gcode\bedlevel\G42.cpp.o
Marlin\src\feature\babystep.cpp: In static member function 'static void Babystep::step_axis(AxisEnum)':
Marlin\src\feature\babystep.cpp:53:45: warning: '--' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
   53 |     if (curTodo > 0) steps[BS_AXIS_IND(axis)]--; else steps[BS_AXIS_IND(axis)]++;
      |                      ~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\feature\babystep.cpp:53:78: warning: '++' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
   53 |     if (curTodo > 0) steps[BS_AXIS_IND(axis)]--; else steps[BS_AXIS_IND(axis)]++;
      |                                                       ~~~~~~~~~~~~~~~~~~~~~~~^
In file included from C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\cores\arduino/Arduino.h:7,
                 from Marlin\src\feature\../sd/../inc/../HAL/../HAL/GD32_MFL/../shared/Marduino.h:36,
                 from Marlin\src\feature\../sd/../inc/../HAL/../HAL/GD32_MFL/HAL.h:27,
                 from Marlin\src\feature\../sd/../inc/../HAL/HAL.h:30,
                 from Marlin\src\feature\../sd/../inc/MarlinConfigPre-5-post.h:29,
                 from Marlin\src\feature\../sd/../inc/MarlinConfigPre-6-type.h:24,
                 from Marlin\src\feature\../sd/../inc/MarlinConfig.h:28,
                 from Marlin\src\feature\../sd/cardreader.h:28,
                 from Marlin\src\feature\binary_stream.cpp:27:
C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\cores\arduino/CoreHandler.h:10:25: error: expected unqualified-id before numeric constant   10 | #define VERSION_MAJOR   1       // Major version
      |                         ^
Marlin\src\feature\binary_stream.h:197:25: note: in expansion of macro 'VERSION_MAJOR'
  197 |   static const uint16_t VERSION_MAJOR = 0, VERSION_MINOR = 1, VERSION_PATCH = 0, TIMEOUT = 10000, IDLE_PERIOD = 1000;
      |                         ^~~~~~~~~~~~~
In file included from Marlin\src\feature\binary_stream.cpp:28:
Marlin\src\feature\binary_stream.h: In static member function 'static void SDFileTransferProtocol::idle()':
Marlin\src\feature\binary_stream.h:140:27: error: 'IDLE_PERIOD' was not declared in this scope
  140 |       idle_timeout = ms + IDLE_PERIOD;
      |                           ^~~~~~~~~~~
Marlin\src\feature\binary_stream.h: In static member function 'static void SDFileTransferProtocol::process(uint8_t, char*, uint16_t)':
Marlin\src\feature\binary_stream.h:146:35: error: 'TIMEOUT' was not declared in this scope
  146 |     transfer_timeout = millis() + TIMEOUT;
      |                                   ^~~~~~~
Marlin\src\feature\binary_stream.h:146:35: note: suggested alternatives:
In file included from C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\system\MFL\MFL\Source\FMC/FMC.hpp:24,
                 from C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\cores\arduino/mfl_api.h:47,
                 from C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\cores\arduino/Arduino.h:10:
C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\system\MFL\MFL\Source\FMC/fmc_config.hpp:280:5: note:   'fmc::FMC_Error_Type::TIMEOUT'   
  280 |     TIMEOUT
      |     ^~~~~~~
In file included from C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\system\MFL\MFL\Source\I2C/I2C.hpp:25,
                 from C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\cores\arduino/mfl_api.h:53:
C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\system\MFL\MFL\Source\I2C/i2c_config.hpp:318:5: note:   'i2c::I2C_Error_Type::TIMEOUT'   
  318 |     TIMEOUT,
      |     ^~~~~~~
In file included from C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\system\MFL\MFL\Source\USART/USART.hpp:25,
                 from C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\cores\arduino/mfl_api.h:68:
C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\system\MFL\MFL\Source\USART/usart_config.hpp:315:5: note:   'usart::USART_Error_Type::TIMEOUT'
      |     ^~~~~~~
Marlin\src\feature\binary_stream.h: At global scope:
C:\Users\Lazarov\.platformio\packages\framework-arduino-mfl\cores\arduino/CoreHandler.h:10:25: error: expected unqualified-id before numeric constant   10 | #define VERSION_MAJOR   1       // Major version
      |                         ^
Marlin\src\feature\binary_stream.h:449:84: note: in expansion of macro 'VERSION_MAJOR'
  449 |   static const uint16_t PACKET_MAX_WAIT = 500, RX_TIMESLICE = 20, MAX_RETRIES = 0, VERSION_MAJOR = 0, VERSION_MINOR = 1, VERSION_PATCH = 0;  
      |                                                                                    ^~~~~~~~~~~~~
Compiling .pio\build\GD32F303RE_creality_mfl\src\src\gcode\bedlevel\M420.cpp.o
Compiling .pio\build\GD32F303RE_creality_mfl\src\src\gcode\bedlevel\ubl\G29.cpp.o
Compiling .pio\build\GD32F303RE_creality_mfl\src\src\gcode\bedlevel\ubl\M421.cpp.o
Compiling .pio\build\GD32F303RE_creality_mfl\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\GD32F303RE_creality_mfl\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\GD32F303RE_creality_mfl\src\src\gcode\config\M220.cpp.o
*** [.pio\build\GD32F303RE_creality_mfl\src\src\feature\binary_stream.cpp.o] Error 1
============================================================ [FAILED] Took 21.58 seconds ============================================================

Environment              Status    Duration
-----------------------  --------  ------------
GD32F303RE_creality_mfl  FAILED    00:00:21.577
======================================================= 1 failed, 0 succeeded in 00:00:21.577 =======================================================

ellensp, 2025-09-15:
Thnkyhead added a version tag so its now marlinfirmware/U8glib-HAL@0.5.5 and Marlin has been updated to use it

closed 2025-09-15 by ellensp