MarlinFirmware/Marlin #27777: [BUG] Marlin (mega2560) freezes/resets after homing while printing

m4rc0d1, 2025-04-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I recompiled (platformio, mega2560) a local copy of bugfix-2.1.x and suddenly my printer resets itself just after the first move of the print (bed heats, hotend heats, homing, first move to the print location and then reset/freeze). Printing via USB (Pronterface) is a little better but the print stops alter the first layer or so.
My bugfix-2.1.x version is dated 2024-07-30 and I uploaded if several times in the past, last time was on March 10. I'm currently using the same configuration used for the working version. I also tried to download the latest bugfix and updated platformio but I get the same outcome.
I get the printer up working again getting the firmware.hex compiled on March 10 from a backup and uploading using avrdude. This I guess should rule out any hardware issue (power, thermal, connectors, etc).
It seems that in the last three weeks something changed in how the firmware is compiled and now the previous working configuration is not working anymore.
Printing from SD the printer sometimes just resets itself (welcome and then main screen), other times it freezes with a "DBUG" led blinking. Using Pronterface an error "KILL caused by KILL button/pin" is received via USB, but there is no kill button on this board.

### Bug Timeline

New bug: the printer works with a version compiled on March 10 and doesn't work with the same source/config compiled on April 1

### Expected behavior

The printer completes the job

### Actual behavior

The printer resets/freezes at the start of the print

### Steps to Reproduce

1. Compile bugfix-2.1.x with the current config
2. Upload firmware 
3. Start any print from SD

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Core XY (modified Ender 4)

### Electronics

Creality Ender 4 (mega2560, RAMPS compatible)

### LCD/Controller

Creality Ender 4 (REPRAP_DISCOUNT_SMART_CONTROLLER compatible)

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/user-attachments/files/19571393/config.zip)

ellensp, 2025-04-03:
the kill button is part of the REPRAP_DISCOUNT_SMART_CONTROLLER LCD  (ie the reset button) 
        #define KILL_PIN             EXP2_08_PIN    (41)

you can try and explicitly disable the kill pin by adding the following to your Configuration.h 
#define KILL_PIN -1

m4rc0d1, 2025-04-03:
I've tried the supplied updated config and disabled the kill pin even if the button is not physically present. Still the printer resets itself after the first move when printing from SD. Printing via USB works for some layers and then the printer freezes and a power cycle is required in order to reset it.

I have backups from the past weeks and what bothers me is that a previously compiled firmware.hex (loaded via avrdude) works with no issue, but the corresponding sources/configs recompiled today don't work.

closed 2025-07-17 by github-actions[bot]