MarlinFirmware/Marlin #27783: [BUG] (bug summary):   Auto Build Marlin will not build for board: BOARD_BTT_OCTOPUS_V1_1 STM32F446ZE_btt BigTreeTech Octopus v1.1 (STM32F446ZE)

KevinMRyan1903, 2025-04-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm attempting to load Marlin onto BigTreeTech Octopus v1.1 (STM32F446ZE) board but when I use the "Auto Build Marlin" I keep getting the error message below.

I tried building other boards like BOARD_BTT_OCTOPUS_PRO_V1_1 which uses the STM32H723ZE chip, and that works. So I think I'm close.

I checked Marlin-bugfix-2.1.x\buildroot\share\PlatformIO\boards folder and it has marlin_BTT_Octopus_v1.json.  So I tried building that one but it doesn't work either.
Then I tried 5 or 6 other STM23F4 Boards and none worked.
So I think maybe it has something to do with that F4 chip board manager?  I can't figure it out.

Any thoughts??

Build error message for BOARD_BTT_OCTOPUS_V1_1:
ms\Projects\Rhino 2M 3D Printer\Driver Board\Marlin-bugfix-2.1.x>
platformio run --silent -e STM32F446ZE_btt
UnknownPlatform: Unknown development platform 'PackageItem <path=C:\Users\15183\.platformio\platforms\ststm32@12.1.1 metadata=PackageMetadata <type=platform name=ststm32 version=12.1.1 spec={'owner': 'platformio', 'id': 8020, 'name': 'ststm32', 'requirements': None, 'uri': None}'

### Bug Timeline

new bug (for me)

### Version of Marlin Firmware

marlin 2.1.x

### Electronics

BigTreeTech Octopus v1.1 (STM32F446ZE)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/19622763/Configuration.zip)

thisiskeithb, 2025-04-06:
All PIO environments build fine for this board on the latest `bugfix-2.1.x`:

[image]
<br />
<br />
<br />
Try following the steps outlined in the "[PlatformIO Errors](https://marlinfw.org/docs/basics/troubleshooting.html#platformio-errors)" section of our troubleshooting guide.

closed 2025-04-06 by thisiskeithb

KevinMRyan1903, 2025-04-06:
still having same issue.
I also tried using the beta version, same issue. 

[image]