MarlinFirmware/Marlin #27792: [BUG] Marlin Simulator (Windows) not working

classicrocker883, 2025-04-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin Simulator isn't compiling.
I'm not sure what changed, this can be just on my end, but even on a whole new install with VSCode and everything I get this:

```
Building in debug mode
Compiling .pio\build\simulator_windows\debug\src\src\HAL\NATIVE_SIM\HAL.o
Marlin\src\HAL\NATIVE_SIM\HAL.cpp:43:12: fatal error: HAL.h: No such file or directory

*************************************************************
* Looking for HAL.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:HAL.h"
* Web  > https://registry.platformio.org/search?q=header:HAL.h
*
*************************************************************

   43 |   #include "HAL.h"
      |            ^~~~~~~
*** [.pio\build\simulator_windows\debug\src\src\HAL\NATIVE_SIM\HAL.o] Error 1
Marlin\src\HAL\NATIVE_SIM\pinsDebug.cpp:25:10: fatal error: ../../inc/MarlinConfig.h: No such file or directory
   25 | #include "../../inc/MarlinConfig.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\simulator_windows\debug\src\src\HAL\NATIVE_SIM\pinsDebug.o] Error 1
*** [.pio\build\simulator_windows\debug\src\src\HAL\shared\Delay.o] Error 1
Marlin\src\HAL\shared\MinSerial.cpp:22:10: fatal error: MinSerial.h: No such file or directory

*******************************************************************
* Looking for MinSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:MinSerial.h"
* Web  > https://registry.platformio.org/search?q=header:MinSerial.h
*
*******************************************************************

   22 | #include "MinSerial.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\simulator_windows\debug\src\src\HAL\shared\HAL.o] Error 1
*** [.pio\build\simulator_windows\debug\src\src\HAL\shared\MinSerial.o] Error 1
```

### Bug Timeline

recently

### Expected behavior

successful compile

### Actual behavior

wont compile, give errors

### Steps to Reproduce

1. **platformio.ini** - `default_envs = simulator_windows`
2. use configs or `#define MOTHERBOARD BOARD_SIMULATED`
3. build

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/user-attachments/files/19718331/configs.zip)

I could be missing something, like some package to install, but compiling normally without simulator works fine.

ellensp, 2025-04-12:
I cannot replicate

New Windows 11 install
install MSYS2 + required packages
install python3 
install git
git clone https://github.com/MarlinFirmware/Marlin.git
install vscode
start vscode 
install platformio and auto build marlin
open up marlin in vscode
open terminal and run  "python .\buildroot\bin\use_example_configs Simulator" to grab simulator example configs
open up ABM and click build

Builds without errors.

ellensp, 2025-04-12:
HAL.h   is a marlin supplies header file https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/HAL/NATIVE_SIM/HAL.h

It cannot be missing without someone deleting it.

closed 2025-04-12 by ellensp

classicrocker883, 2025-04-13:
thanks for trying. that is the odd thing - i *do* have those header files, so I don't get why its showing that. see I was able to use WSL on Windows 11 using `simulator_linux_release` using the same repo and it worked well. that's what I dont get.

I can reinstalling some preliminary things i wasnt sure which ones so thanks to laying those out