MarlinFirmware/Marlin #27793: [BUG] Tronxy Chitu V6 X_STOP_PIN PG10 does not work

taxidriver1977, 2025-04-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After an update to version 2.1.2.5, the X_min pin no longer works. The last version that was installed is Bugfix2.0.x.

This bug also seems to be on the current bugfix 2.1.x.

The board is a Chitu V6 from Tronxy

### Expected behavior

Troubleshooting for X_STOP_PIN PG10
Currently PG10 is not usable

### Actual behavior

PG10 X_STOP_PIN ist nor usable.

Current temporary solution:
X_STOP_PIN change from PG10 to Z_Max port PA13

### Steps to Reproduce

This bug is also on latest und bugfix.
 

### Version of Marlin Firmware

2.1.2.5

### Printer model

Tronxy x5sa

### LCD/Controller

Tronxy

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[chitu V6 Bug Config Files.zip](https://github.com/user-attachments/files/19720909/chitu.V6.Bug.Config.Files.zip)

Configuration.h and Configuration_adv.h
pins_CHITU3D_common.h and pins_CHITU3D_V6.h

Mazecreator, 2025-04-16:
I had a similar problem, what I found is changing the pin as follows: (Mine was PA14 for the BETA2, but change it to PG9)
//#define Z_STOP_PIN                          PA14
#define Z_STOP_PIN                          PG9

taxidriver1977, 2025-04-17:
This is not really a bug.

Simply deactivate the line Zmin_Probe_Use_Z_Min_Endstop_Pin and the z_minprobe_pin PG9 is used.

//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN PG9

closed 2025-07-31 by github-actions[bot]