MarlinFirmware/Marlin #27795: [BUG] GD32F303RE_creality_mfl fails to boot

ellensp, 2025-04-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

more info to come

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 3 pro

### Electronics

Creality V422 WITH gd32f303

### LCD/Controller

CR10_STOCKDISPLAY

### Other add-ons

BLTOUCH

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

Other (explain below)

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

bmourit, 2025-04-14:
Yes, I'll look into this.

bmourit, 2025-04-14:
> Yes I tried that also

Okay, I'll still see what else is up.

ellensp, 2025-04-14:
Yes, non boot is a PITA, which i why I deferred to you, someone with a greater understanding of the low level code. 
I don't even know where my stlink is hiding at present..

ellensp, 2025-04-14:
#define SERIAL_DMA seems to have made no difference

bmourit, 2025-04-14:
Okay, thanks for trying. I'll be at my machine in a moment to debug it properly.

bmourit, 2025-04-14:
Alright, I built with your config files, except with SERIAL_PORT set to 0 and BAUDERATE 115200, and it is booting fine for me. The only thing my board has that is different than yours is - Yours has BLTOUCH and Your drivers are TMC2208 mine are A4988. I do recommend setting enabling SERIAL_DMA and ENDSTOP_INTERRUPT_FEATURE, but I was still able to boot without those..

So, this may seem like a dumb question, but do you for sure have the GD32F3034RET6 chip and not GD32F103RET6. IIRC, some people reported getting the GD32F103RET6 chip, which does not have FPU and may explain the issues.

ellensp, 2025-04-15:
I have 5 of these boards in a variety of configurations for testing. 

I am using the V4.2.2 with the GD32F303ET6, as I am replicating what another user did on discord, with the same none booting result.

I do not know what stepper drivers this board actually has, I wasn't worried  about moving and the steppers.  
TMC2208_STANDALONE vs A4988 are almost identical, only timing changes 

I started with stock Marlin config for a  Creality Ender 3 pro + bltouch  https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20Pro/CrealityV427%20-%20BLTouch and changed the Motherbaord to a BOARD_CREALITY_V422_GD32_MFL 

Also re 4.2.7, Also that is minimally different.  Just Step and direction IO pins swapped over for each axis, nothing that should stop booting (I have helped coutless people over the years with "Stepper wont move" because they installed V422 firmware on a V427 or vice versa board)

ellensp, 2025-04-15:
Updated to todays bugfix

Reduced config down to just
#define MOTHERBOARD BOARD_CREALITY_V422_GD32_MFL
#define SERIAL_PORT 0
#define BAUDRATE 115200
#define FAN_SOFT_PWM
 
platformio.ini 
[platformio] default_envs = GD32F303RE_creality_mfl
[env] monitor_speed     = 115200

Still no boot.

Here is my full build logs, please verify my system is getting correct versions of required files.

[build.zip](https://github.com/user-attachments/files/19747085/build.zip)

Not I resinatlled GD-Ender-3 ProHW4.2.2SW2.0.8.2CRTouchFilamentEuropeMulti  from creality and it booted fine.

bmourit, 2025-04-15:
Just for the sake of being thorough, would you delete the platformio/packages/framework-arduino-mfl so that it redownloads it again (you probably have already done this, but just in case). Otherwise, I'll have to dig a little deeper here. 😖

bmourit, 2025-04-15:
> I have 5 of these boards in a variety of configurations for testing.
> 
> I am using the V4.2.2 with the GD32F303ET6, as I am replicating what another user did on discord, with the same none booting result.
> 
> I do not know what stepper drivers this board actually has, I wasn't worried about moving and the steppers. TMC2208_STANDALONE vs A4988 are almost identical, only timing changes
> 
> I started with stock Marlin config for a Creality Ender 3 pro + bltouch https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20Pro/CrealityV427%20-%20BLTouch and changed the Motherbaord to a BOARD_CREALITY_V422_GD32_MFL
> 
> Also re 4.2.7, Also that is minimally different. Just Step and direction IO pins swapped over for each axis, nothing that should stop booting (I have helped countless people over the years with "Stepper wont move" because they installed V422 firmware on a V427 or vice versa board)

You have a bunch of these boards? Im in the process of trying to make sure I get GD32 ones. You might be my new best friend.

ellensp, 2025-04-15:
I had not nuked platformio/packages/framework-arduino-mfl   I first tried this on the 11 April .

After nuking this it does boot. no lcd visible yet, but trying a few things

ellensp, 2025-04-15:
closing issue as it now boots. but will continue discussion around the LCD

closed 2025-04-15 by ellensp

bmourit, 2025-04-16:
You have no idea how much of a relief that is. Also, is this the stock 12964 LCD?

ellensp, 2025-04-16:
yes a stock 128x64 mono lcd CR10_STOCKDISPLAY.

ellensp, 2025-04-18:
I've had a look around, what I see is the following.

the lcd is using hardware spi

PIN_SPI_MOSI    PB15
PIN_SPI_MISO    PB14
PIN_SPI_SCK     PB13

On STM32F103 this is SPI2, but on GD32F303 this is SPI1
 
In 
https://github.com/bmourit/ArduinoCore-MFL/blob/main/variants/MFL_CREALITY_422/variant.h

I see the following

```
// SPI Definitions
#define SUPPORT_SPI0
#define PIN_SPI_SS      PA4
#define PIN_SPI_MOSI    PA7
#define PIN_SPI_MISO    PA6
#define PIN_SPI_SCK     PA5

```
So these doesnt seem correct.

I tried updating these pins, but i got nowhere.

bmourit, 2025-04-18:
@ellensp Do you mind trying it after applying these two patches?
https://github.com/MarlinFirmware/Marlin/pull/27800
https://github.com/MarlinFirmware/Marlin/pull/27801

ellensp, 2025-04-18:
Already did that before I started looking into this today.  Just in case they changed anything... 
It didn't.

ellensp, 2025-04-18:
I have display!

Once you pointed out it was software spi (I was certain it was hardware, I clearly messed up!)  

In Marlin/src/inc/Conditionals-2-LCD.h is
```
#elif ENABLED(CR10_STOCKDISPLAY)

  #define IS_RRD_FG_SC 1
  #define NO_LCD_SDCARD

  #define LCD_ST7920_DELAY_1           125
  #define LCD_ST7920_DELAY_2           125
  #define LCD_ST7920_DELAY_3           125

```

I upped these to 150 and I get display..

As you probably know these set the software spi timing 

I then reverted that and set it at the board level adding it to pins_CREALITY_V422_GD32_MFL.h
with

#define BOARD_ST7920_DELAY_1 150
#define BOARD_ST7920_DELAY_2 150
#define BOARD_ST7920_DELAY_3 150

 Which also works.

I'm guessing my cr10_stockdisplay is slower than yours (they where not all identical) 
and the bump from 72mhz to 120mhz was enough to push it out of tolerance.

bmourit, 2025-04-20:
Under GD32_MFL in HAL.h we have:

// Default graphical display delays
#define CPU_ST7920_DELAY_1 300
#define CPU_ST7920_DELAY_2  40
#define CPU_ST7920_DELAY_3 340

We should increase these here to a higher level to make sure we properly support the LCDs. Do you mind playing with these for your LCD? The lowest possible is what we want to go for, then we should increase it for all boards here as well.

bmourit, 2025-04-21:
Looking at the code base, the driver is taking the _MAX of LCD_ST7920_DELAY_x, BOARD_ST7920_DELAY_x, CPU_ST7920_DELAY_x. Our current values of 300, 40, and 340 are carried over from STM32, though the 40 value really is meaningless here since it wont be less that 125. 

Maybe we can do:
#define CPU_ST7920_DELAY_1 300
#define CPU_ST7920_DELAY_2 150
#define CPU_ST7920_DELAY_3 340

In the HAL.h file, unless we need different values for another board (Ie. Aquila with GD32F103 @ 108MHz).
This would save duplicating code in each pin file.

ellensp, 2025-04-21:
I just removed my BOARD_ST7920_DELAY_X  and it still  works.

I have not changed CPU_ST7920_DELAY_X or LCD_ST7920_DELAY_X

I added a lcd menu and gcode (M2000, just needed a unused gcode)  to display (and set) the software api delays and it shows the expected result

03:12:00.183 > echo:; Software SPI settings:
03:12:00.186 > echo:  M2000 L300 S125 H340

I have no idea what is going on any more..  
I think we should just forget this...  till it happens again

bmourit, 2025-04-21:
Actually, this is very interesting. When I first started this project I remember having to increase delay as well otherwise I had screen garble. But then later on the default values worked again. So strange! This is really only changing the CPU_ST7920_DELAY_2 value from the default of 125, to 150. So not to much. My LCD seems a little more stable with this value. 
If you notice any garbling or tearing maybe we should just bump it up. It is still much lower than most other platforms.

Let's give it a few days of testing. I have a draft PR open, but I can always close it if we are good.

Also, thats a great mod!