MarlinFirmware/Marlin #27802: [FR] After Dual Axis Homing, do BLTOUCH homing, to increase precision.

12oclocker, 2025-04-17:
### Describe the feature you want

Machines such as AnyCubic i3 Mega, and others, have independent Z motors with their own homing sensors for each Z motor. And no sync belt. To increase accuracy for these types of machines, it would be nice to have an option such as...
#define BLTOUCH_HOMING_AFTER_DUAL_Z_HOMING
Which would allow the homing of the two Z axis motors with their own homing sensors to get them in sync; then we do homing again but ignore the Z homing sensors and perform the homing with BLTOUCH. This will allow repeatable accuracy printing at any BED temperature, since BLTOUCH would home to the bed surface after Z motors are in sync.
Many users of dual axis machines would benefit from this feature.

ellensp, 2025-04-18:
see Z_STEPPER_AUTO_ALIGN

12oclocker, 2025-04-18:
I am aware of the ability To replace independent z homing sensors with BLTOUCH, the BL touch sensor will move to the left side of the bed and the right side of the bed repeatedly and perform an alignment of the dual z motors. I have tried it, but it is very slow. The method described in the feature request is a much faster method for accurate homing.

Nuck-TH, 2025-04-19:
You will need to make meshes for different temperatures anyway and doing them has same effect, so this would be redundant.

12oclocker, 2025-04-28:
Trying to use BLTOUCH instead of the endstop switches yeilds terrible accuracy on my AnyCubic i3 Mega, I have tried it. AnyCubic i3 Mega uses a Dual Z motor and dual Z endstop switches, trying to use Z_STEPPER_AUTO_ALIGN and BLTOUCH to home dual Z motors yields terrible accuracy, even with 10 attempts the machine is not as accuracy as using the dual endstop switches, and it takes much longer. The BLTOUCH is too far away from the motors, the enstop switches are right under the Z screw.
Also I think the feature may help with this problem I am experiencing... https://github.com/MarlinFirmware/Marlin/issues/27680

12oclocker, 2025-04-29:
After testing the Z_STEPPER_AUTO_ALIGN on AnyCubic i3 Mega S, I have found the accuracy is not good, cannot achieve better than 0.3mm, and the reason is because the algorithm does not account for the spacing of the Z1 and Z2 motors, their is no where to enter that info in the configs that I could find (maybe I am missing it?).  Distance multiplied by tan(angle) = height ... the common point of the angle is the center of the Z1 axis if you are going to adjust the Z2, and the common point of the angle is Z2 if you are going to adjust Z1. The two probe points can be assumed they spread even from center of the bed, but a datapoint for that would increase accuracy. The probe points can then be used to calculate what the error is from the common angle point and how much to move Z1 or Z2, and actually do it with fairly good accuracy without needing a bunch of retries.