MarlinFirmware/Marlin #27805: [BUG] Reversed Move Axis adjustment screen direction: The 3 reverse options in Configuration.h don't help. The Dream Maker Overlord Pro setup files have the same problem/bug.

a4jp-com, 2025-04-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

[image]

My stepper motors are functioning properly and move the correct way through G-Code in Pronterface and Cura.
The buttons on the Dream Maker Overlord Pro move the selection up and down on the main menu correctly. 
Everything seems great and is moving the correct way through G-Code and any slicer I use. Even the Arduino IDE.

Here is the problem: The Move Axis screens are reversed. X, Y and Z. 

The 3 options to reverse just the menu or motors don't fix this problem either.

Can you please fix the code for the Dream Maker Overlord Pro or any printer with this problem?

[Marlin-Bugfix-2-1-X-Version-Dream-Maker-Overlord-Pro.zip](https://github.com/user-attachments/files/19814291/Marlin-Bugfix-2-1-X-Version-Dream-Maker-Overlord-Pro.zip)

### Bug Timeline

Every since trying to update to Marlin 2 (about 2 months)

### Expected behavior

I push up and the menu selector moves up.
I push down and the menu selector moves down.
I type in G-Code to move the nozzle and it moves the correct way.
I use Pronterface or Cura to move the nozzle with buttons and the nozzle moves the correct way.

If something is moving the wrong way there should be a line of code to define to fix the problem in the Configuration.h file. There isn't for this situation.

### Actual behavior

The problem: 
Only the Move Axis screen is moving the nozzle in the reverse direction.
Up is down; forward is reverse; left is right. This is only on the Move Axis screens. The other screens are all good.

I tried all 3 versions of reversing code in the Configuration.h file but it doesn't help. 

--------------------------------------------------------------------

Motion > Auto Home

Move Axis > Move Z > 
Move 10mm > push up or down. The direction is wrong.
Move 1.0mm > push up or down. The direction is wrong.
Move 0.1mm > push up or down. The direction is wrong.
Move 0.025mm > push up or down. The direction is wrong.

Move Axis > Move X > 
Move 10mm > push up or down. The direction is wrong.
Move 1.0mm > push up or down. The direction is wrong.
Move 0.1mm > push up or down. The direction is wrong.
Move 0.025mm > push up or down. The direction is wrong.

Move Axis > Move Y > 
Move 10mm > push up or down. The direction is wrong.
Move 1.0mm > push up or down. The direction is wrong.
Move 0.1mm > push up or down. The direction is wrong.
Move 0.025mm > push up or down. The direction is wrong.

### Steps to Reproduce

Install the latest version of Marlin Bugfix 2.1.x firmware

Motion > Auto Home  

Then click Move Axis > 

Move X
Move Y
Move Z

Move 10mm
Move 1.0mm
Move 0.1mm
Move 0.25mm

Example: the problem is on the screen where it shows this
Move Z: +228.4

----------------------
Turn on the printer

Connect with Pronterface

G28

Motion > Auto Home  

Then click Move Axis > 

Move X
Move Y
Move Z

Move 10mm
Move 1.0mm
Move 0.1mm
Move 0.25mm

Example: the problem is on the screen where it shows this
Move Z: +228.4
-------------------------
Menu > Level Bed

Enter to start

Move Z: 0.000

Same problem here.

### Version of Marlin Firmware

Marlin Bugfix 2.1.x

### Printer model

Dream Maker Overlord Pro V1.1

### Electronics

MCU ATMega 2560 - Dream Maker Overlord V1.1 board

### LCD/Controller

Overlord OLED display/controller with i2c buzzer and LEDs  #define OVERLORD_OLED 128x64

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I started off trying to fix this in Arduino IDE and then updated to 2.3.5.
I'm now compiling with Visual Studio Code 64x V. 1.99.3 and PlatformIO Core 6.1.18 Home 3.4.4

The Overlord OLED Screen (128x64) 
`#define OVERLORD_OLED
#if ENABLED(OVERLORD_OLED)
  #define IS_ULTIPANEL 1
  #define U8GLIB_SH1106
  /**
   * PCA9632 for buzzer and LEDs via i2c
   * No auto-inc, red and green leds switched, buzzer
   */
  #define PCA9632
  #define PCA9632_NO_AUTO_INC
  #define PCA9632_GRN         0x00
  #define PCA9632_RED         0x02
  #define PCA9632_BUZZER
  #define PCA9632_BUZZER_DATA { 0x09, 0x02 }
  #define STD_ENCODER_PULSES_PER_STEP     1 // Overlord uses buttons
  #define STD_ENCODER_STEPS_PER_MENU_ITEM 1
#endif`

Extruder for OverLord 3D Printer
Motor: 17PM-FA38CN01CN
T15407-01
Minebea Co. Ltd.
Thailand
3.6V 2.0A
1.8 Degrees

X, Y and Z Motors
17PM-K446-04VS
T15327-01
Minebea Co. Ltd.
Thailand
4.6V 1.1A
200 Steps 1.8 Degrees +/- 0.09 Degrees
3.92kg*cm

Stepper Controller: Custom 17 pin PCB with DRV8825 4ATG5 CH4H

[elecrical 20160420115657wl6ipo.pdf](https://github.com/user-attachments/files/19814243/elecrical.20160420115657wl6ipo.pdf)
[English electrical diagram 20160420115657wl6ipo.pdf](https://github.com/user-attachments/files/19814242/English.electrical.diagram.20160420115657wl6ipo.pdf)

a4jp-com, 2025-04-19:
@thinkyhead / @ellensp I found the problem. 

In the src/lcd/menu/menu_motion.cpp file the float was positive instead of negative. 

Line 70
`const float diff = float(int32_t(ui.encoderPosition)) * ui.manual_move.menu_scale;`

```
// Get the new position
    const float diff = -float(int32_t(ui.encoderPosition)) * ui.manual_move.menu_scale; 
    //Reverses the X, Y & Z direction of the menu motion direction
    (void)ui.manual_move.apply_diff(axis, diff, min, max);
    ui.manual_move.soon(axis);
    ui.refresh(LCDVIEW_REDRAW_NOW);
```

This is how I fixed it:
`const float diff = -float(int32_t(ui.encoderPosition)) * ui.manual_move.menu_scale;
`
But can you make some define conditions to reverse the motion direction for people with printers like mine that come across the same problem? Setting to reverse all X, Y, Z and E or just the ones we want individually.

New options to fix any direction
//#define REVERSE_MENU_MOTION_DIRECTION
//#define REVERSE_MENU_MOTION_DIRECTION_X
//#define REVERSE_MENU_MOTION_DIRECTION_Y
//#define REVERSE_MENU_MOTION_DIRECTION_Z
//#define REVERSE_MENU_MOTION_DIRECTION_E

These are nice when we wanna switch everything though.
//#define REVERSE_ENCODER_DIRECTION
//#define REVERSE_MENU_DIRECTION
//#define REVERSE_SELECT_DIRECTION
They just don't work in my case on the Dream Maker Overlord Pro.

thinkyhead, 2025-04-20:
> I tried _**each one**_
> I tried all 3 versions of reversing code in the `Configuration.h` file but it doesn't help.

From your description it looks like you want to define both `REVERSE_ENCODER_DIRECTION` and `REVERSE_MENU_DIRECTION` at the same time.
```cpp
#define REVERSE_ENCODER_DIRECTION
#define REVERSE_MENU_DIRECTION
```
The reversed encoder direction will make everything *except* menus inverted. And then `REVERSE_MENU_DIRECTION` will fix the reversed menu direction caused by the first option.