MarlinFirmware/Marlin #27807: [BUG] USB flash drive support broken

DaLiV, 2025-04-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

commit 0f7fb5d88df2ff450b6f7426c55ac0a979a5ed55 - breaks USB drive accessiblity

### Expected behavior

Acessible both USB and SD cards

### Actual behavior

USB-Drive not more accessible

### Steps to Reproduce

1. apply provided config from test.zip into git source (after 0f7fb5d88df2ff450b6f7426c55ac0a979a5ed55)
2. pio run
3. if test on device - no USB accessible, only SD content . 

as alternative to test on device can also see that against previous commit resulting filesize will be 10Kb less ...
(can treat as indirect sign of not compiled-in code block)
(1.) compiled on commit 428f721cbfc040ca9e57abd590c3798ed719fd8d
Flash: [==        ]  16.0% (used **167948** bytes from 1048576 bytes)
(1.) compiled on commit 0f7fb5d88df2ff450b6f7426c55ac0a979a5ed55
Flash: [==        ]  15.1% (used 158044 bytes from 1048576 bytes)

### Version of Marlin Firmware

git-bugfix 2.1.x

### Electronics

BTT Octopus Pro V1.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[test.zip](https://github.com/user-attachments/files/19821921/test.zip)

DaLiV, 2025-04-19:
if forced definition from Configuration.h for
#define MULTI_VOLUME
#define USB_FLASH_DRIVE_SUPPORT

then seems works ... but that configurations are initially placed in Configuration_adv.h, along with other conditionals . 
seems that file not always nicely taken in compilation (as in this case)

DaLiV, 2025-04-19:
seems need in
Conditionals-4-adv.h instead of Conditionals-2-LCD.h

--- a/Marlin/src/inc/Conditionals-2-LCD.h
+++ b/Marlin/src/inc/Conditionals-2-LCD.h
@@ -33,15 +33,15 @@
 //========================================================
.
 // Support for SD Card and other file storage
-#if ENABLED(SDSUPPORT)
-  #define HAS_MEDIA 1
-#endif
-#if ENABLED(MULTI_VOLUME)
-  #define HAS_MULTI_VOLUME 1
-#endif
-#if ENABLED(USB_FLASH_DRIVE_SUPPORT)
-  #define HAS_USB_FLASH_DRIVE 1
-#endif
+//#if ENABLED(SDSUPPORT)
+//  #define HAS_MEDIA 1
+//#endif
+//#if ENABLED(MULTI_VOLUME)
+//  #define HAS_MULTI_VOLUME 1
+//#endif
+//#if ENABLED(USB_FLASH_DRIVE_SUPPORT)
+//  #define HAS_USB_FLASH_DRIVE 1
+//#endif
.
 //
 // Serial Port Info

--- a/Marlin/src/inc/Conditionals-4-adv.h
+++ b/Marlin/src/inc/Conditionals-4-adv.h
@@ -31,7 +31,6 @@
 //
 #include "MarlinConfigPre-4-adv.h"
 //========================================================
-
 #if ENABLED(MARLIN_SMALL_BUILD)
   #undef EEPROM_CHITCHAT
   #undef CAPABILITIES_REPORT
@@ -1494,3 +1493,13 @@
   #undef AUTO_REPORT_SD_STATUS
   #define AUTO_REPORT_SD_STATUS
 #endif
+
+#if ENABLED(SDSUPPORT)
+  #define HAS_MEDIA 1
+#endif
+#if ENABLED(MULTI_VOLUME)
+  #define HAS_MULTI_VOLUME 1
+#endif
+#if ENABLED(USB_FLASH_DRIVE_SUPPORT)
+  #define HAS_USB_FLASH_DRIVE 1
+#endif

thisiskeithb, 2025-04-19:
> seems need in
> Conditionals-4-adv.h instead of Conditionals-2-LCD.h

Please open a Pull Request with the fix.

thisiskeithb, 2025-04-19:
I can confirm that despite `MULTI_VOLUME` & `USB_FLASH_DRIVE_SUPPORT` being enabled, they do not show as such in `Conditionals-2-LCD.h`: https://github.com/MarlinFirmware/Marlin/blob/26b3f0b00f897a0624fd96d064f50cb3cf54d3b2/Marlin/src/inc/Conditionals-2-LCD.h#L39-L44

Moving those conditionals to `Conditionals-4-adv.h` fixes it.

DaLiV, 2025-04-19:
i not fully like that same code must be in Conditionals-2-LCD.h and Conditionals-4-adv.h
therefore think pull request will be not the best way ... 

why LCD not taken in this case not clearly seems
logically that muste be related to Conditionals-4-adv as that meant as initial parser of all that is inside Configuration_adv.h.
Conditionals-LCD seems created for LCD-related mostly.

but additionally - in case if leave that part only in -adv.h and removed from -LCD.h - same problem - USB part not compiled-in also.

P.S. in that dir processing order of files is not so clear/linear () ... seems point for improvement also.

thinkyhead, 2025-04-21:
I also have the TS35 so I was able to do a pretty close test. Based on Keith's input I updated the `bugfix-2.1.x` conditionals. I also updated your configs for the upcoming 2.1.3-b3 so here they are:

[configs-updated.tar.gz](https://github.com/user-attachments/files/19827741/configs-updated.tar.gz)

If you use these to do a new build with the latest code the file navigation now seems to work pretty well (except the cosmetic bug where portions of the screen are not being cleared).

I will see if I can fix those cosmetic issues, and in the meantime give the updated firmware a test and see if it works for you.

thinkyhead, 2025-04-21:
Let us know if the newest code is working, and if so this issue can be closed!

DaLiV, 2025-04-21:
so - tested - seems ok. but partially

seems exist another bug:
when inserted both (SD+USB) - shown menu "select from media -> USB/SD" and on each can be files selected
when USB us taked off (no matter operating or at boot time) and SD card still in  - menu not appears.
si SD card not always accessible
if send over Serial "M21 S" - then SD is shown, 
suppose at point where choice for media availability missing point to check "MULTI_VOLUME" case - check Both.
sd card detection seems not working (insert-taking off not detecting in debug)
prior build has next messages, that now shown not:
```
>>> CardReader::manage_media()  X0.0000 Y0.0000 Z0.0000 A0.0000
Media present: 1 -> 0
echo:SD card released
<<< CardReader::manage_media()  X0.0000 Y0.0000 Z0.0000 A0.0000
>>> CardReader::manage_media()  X0.0000 Y0.0000 Z0.0000 A0.0000
Media present: 0 -> 1
echo:SD card ok
<<< CardReader::manage_media()  X0.0000 Y0.0000 Z0.0000 A0.0000

```
at boot shown:
```
echo:EEPROM Initialized
echo:[3247] thermalManager.init()
echo:[3250] print_job_timer.init()
echo:[3250] endstops.init()
echo:[3250] stepper.init()
echo:[3250] hal.watchdog_init()
echo:[3251] test_tmc_connection()
Testing X connection... OK
Testing Y connection... OK
echo:[3268] setup() completed.
>>> CardReader::manage_media()  X0.0000 Y0.0000 Z0.0000 A0.0000
Media present: 2 -> 0
echo:SD card released
<<< CardReader::manage_media()  X0.0000 Y0.0000 Z0.0000 A0.0000
Starting USB host...started. rev unknown.

```
"SD card released"-however sd card used for firmware update and still in reader  ... and works after "M21 S"

pins debug shows pin and card presence:
```
card in
PIN: PC14        M42 P46          SD_DETECT_PIN                          Input  HIGH
card out
PIN: PC14        M42 P46          SD_DETECT_PIN                          Input  LOW

```
so defined, but not checked/triggering insertion/release in firmware .

thinkyhead, 2025-04-23:
I'm currently taking copious notes on the current behavior of the "SD Card" / "USB Drive" family of menu items, and they are definitely well out of date in terms of how they can and should behave. I've already found a few things and fixed them in #27815 and I will be adding more to that PR as I work out a much more logical arrangement of menu items for the available / inserted / mounted drives. So let's continue this conversation over there!

closed 2025-04-23 by thinkyhead