MarlinFirmware/Marlin #27809: (Bug) Can Adjust 2 out of 5 Extruders

ncc2707, 2025-04-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

To start with, I'm new to all of this. Just finished a retro fit  for the Cyclops 5-in1-out extruder, using BTT Octopus. The display only shows E0 and E1 in the Menu Sets per mm. I need the rest of the Extruders to finish setup. Tried to do this through Marlin and my steps test was, 100 cm for 100mm called for. I used the same numbers from E1 to the others in VS. Can't find it in the code to list more om the display.

### Expected behavior

To be able to adjust all of the extruders that are being used.

### Actual behavior

Can only see E0 and E1.

### Version of Marlin Firmware

Marlin 2.1.2.5 (April 19 2025 22:40:38)

### Printer model

Anycubic 4Max

### Electronics

BigTreeTech Octopus v1.1

### LCD/Controller

BigTreeTech GD TFT35_E3_V3.0

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[ConfigurationFiles 42-20-2025.zip](https://github.com/user-attachments/files/19826047/ConfigurationFiles.42-20-2025.zip)

ncc2707, 2025-04-20:
Thanks for responding so quickly. I did know if it was an on-going type or not. Thanks again and I'll let the madam know. :)

ellensp, 2025-04-21:
[image]

Lcd shows and lets you edit all Extruder steps/mm

ellensp, 2025-04-21:
Your conf is missing some of the  BTT TFT Marlin dependencies see
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/README.md#marlin-dependencies

ncc2707, 2025-04-22:
Thank you so much. I had removed the Ext cables. I saw that you was in Marlin, I felt a little bad. Live and learn!