MarlinFirmware/Marlin #27814: [BUG] (bug summary)z offset problem on tinybee board

akhteh, 2025-04-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In new versions of Marlin, there is a z offset problem in tinybee v1 boards During printing, when we give z offset, it is not applied to the print immediately and has no effect, it is only applied in the next print I have a request to follow up and fix the problem in the new version, thank you
It could be a bug due to the display. I tested it with the btt_mini_12864 v2 display, but many users who have this board have the same problem.

### Bug Timeline

It was apparently correct in version 2.0.9.2, but I could not test it because it was old, but I tested all the later versions and they have this problem.

### Expected behavior

I expect that due to the many users who have the btt_mini_12864 display, the problem will be solved in the new version of Marlin and we will be able to increase and decrease the z offset during printing like other boards..

### Actual behavior

It has no effect when printing with btt_mini_12864 board

### Version of Marlin Firmware

2.1.2.5 

### LCD/Controller

Btt_mini_12864 v2.0 

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/19839877/Marlin.zip)

chebhou, 2025-04-25:
sorry for hijacking this but do you have problem ith the SD card and screen too

akhteh, 2025-04-25:
hello Not only when I use the WebUI and refresh the memory or want to send
a file to the memory via Wi-Fi, both the memory on the board and the memory
I put on the screen, the screen turns off, but when I put the file on the
memory and import it from the screen, I can easily print. But the biggest
problem is the z offset during printing, which has no effect when I set it
during printing, but it is applied in the next print.

closed 2025-08-17 by github-actions[bot]