MarlinFirmware/Marlin #27816: [BUG] "M43 - Debug Pins" clashes with laser cooler temperature

dty717, 2025-04-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I ran M43 E1 and toggled the endpoints of x , the temperature is changing from 20 deg to 40 deg and triggered the laser cooler mini temperature.

### Version of Marlin Firmware

Marlin bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/19858916/Marlin.zip)

ellensp, 2025-04-23:
We need sufficient files to replicate your issue
You have clearly edited pins files to add all the additional stepper drivers.

Please provided all edited files.

dty717, 2025-04-23:
[pins_RAMPS.zip](https://github.com/user-attachments/files/19859019/pins_RAMPS.zip)

ellensp, 2025-04-23:
PINS_DEBUGGING is not even enabled in provided config files.

dty717, 2025-04-23:
sorry for that, because of the bug ,i remove the feature.

dty717, 2025-04-23:
[Marlin.zip](https://github.com/user-attachments/files/19859625/Marlin.zip)

closed 2025-08-06 by github-actions[bot]