MarlinFirmware/Marlin #27821: [BUG] AnyCubic i3 Mega with BLTOUCH, Z Probe Offset has no effect.

12oclocker, 2025-04-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

AnyCubic i3 Mega with BLTOUCH, Z Probe Offset has no effect.
The i3 Mega has a Dual Z motor, with endstops for each motor, it is accurate for alignment of the Z motors to keep using the Z1 and Z2 endstops, for this reason BLTOUCH is not used for homing, only MESH.
When Mesh is created, it is created using the height of probe, Z Probe Offset is NOT subtracted out. 
After Homing and having ENABLE_LEVELING_AFTER_G28 enabled. The Z Probe Offset is STILL not subtracted after homing and after applying a mesh; The nozzle moves around at Z Probe height instead of the bed surface.
I have installed Marlin on my Ender3 and Alphawise U20, both work fine with BLTOUCH.

AnyCubic i3 Mega needs to use it's own Z1 and Z2 endstops instead of using the BLTOUCH for Z homing, which I think that may have something to do with the Z Probe Offset having no effect. Because on my other machines that have a single Z axis and that use the BLTOUCH for homing, the BLTOUCH and MESH works fine.
Tested all 3 machines with 2.0.2.5 and 2.1.3-b2 and bugfix-2.1.x All machines work with BLTOUCH and apply the Z Probe Offset correctly except the AnyCubic i3 Mega with dual Z motors.

### Bug Timeline

Bug exist in 2.0.2.5 and 2.1.3-b2 and bugfix-2.1.x

### Expected behavior

I expect either the Z Probe Offset to be subtracted from the MESH when it is created. Or I expect the Z Probe Offset to be Subtracted from the MESH after homing and when the mesh is being applied to the active print.

### Actual behavior

Z Probe Offset is never applied to anything.

### Steps to Reproduce

[configs.zip](https://github.com/user-attachments/files/19936833/configs.zip)

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

AnyCubic i3 Mega with BLTOUCH and MKS_MINI_12864_V3

### Electronics

with BLTOUCH and MKS_MINI_12864_V3

### LCD/Controller

MKS_MINI_12864_V3

### Other add-ons

BLTOUCH

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2025-04-28:
#define BABYSTEP_GLOBAL_Z               // Combine M424 Z and Babystepping

12oclocker, 2025-04-28:
Hello, enable of BABYSTEP_GLOBAL_Z has no effect, a quick grep search 
find ./Marlin-bugfix-2.1.x -type f -execdir grep -H -n 'BABYSTEP_GLOBAL_Z' {} \;
shows that define only appears in the source code once, Configuration_adv.h:2326:  #if ANY(BABYSTEP_ZPROBE_OFFSET, BABYSTEP_GLOBAL_Z) 
and if I enable that setting, then I get error...
#error "BABYSTEP_HOTEND_Z_OFFSET requires 2 or more HOTENDS.
thus it appears to only be used only to enable BABYSTEP_HOTEND_Z_OFFSET which says it is "For multiple hotends"

Also I have been looking at my 3D printers that have a single Z axis and the BLTOUCH works, It appears the Z Probe Offset is subtracted from the probe points, giving mesh values near zero.
While the AnyCubic machine with dual Z motors, the saved MESH values do NOT have the Z Probe Offset subtracted out.
I now believe this is the problem. Z Probe Offset is not subtracted from saved MESH probe points when using endstop sensor instead of BLTOUCH for homing. Thus USE_PROBE_FOR_Z_HOMING or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN not being enabled is causing the bug of the saved MESH values to NOT have the Z Probe Offset subtracted out.

12oclocker, 2025-04-28:
I finally found a workaround for the bug, 

I have ENABLE_LEVELING_AFTER_G28 enabled, so that my bed leveling MESH will always be applied after homing.

//bug workaround...
#define Z_AFTER_HOMING  0 //must be zero for AutoBedLeveling to work properly using a Z endstop and BLTOUCH
#define EVENT_GCODE_AFTER_HOMING "G92 Z0 \n G1 Z10 F3000" //G92 Z0 is critical for AutoBedLeveling to actually subtract ZProbeOffset during MESH creation when using a BLTOUCH and using a Z endstop
//the "G1 Z10 F3000" is simply to set the Z position after homing, since Z_AFTER_HOMING must be zero.

So if you are using a Z endstop and BLTOUCH, the ZProbeOffset will never be subtracted out of the MESH you create, unless you use the settings above.

So after doing the Z probe wizard to set your Z probe offset. You can now do bed leveling and the MESH will be created and subtract out the Z probe offset correctly from the MESH. And if you ever change your Z Probe Offset, you probably will need to rebuild your mesh again.

12oclocker, 2025-05-28:
update, the bug fix does Not work if your Z homing sensors trigger above the the bed surface. even if you enable negative z movement. A proper bug fix will need implimented in code, because bltouch and z offset and bed leveling does Not work on multi z motor printers that require z endstops.

12oclocker, 2025-07-28:
Bug: Confirmed
This bug still exist. tested with latest bugfix-2.1.x

Z Probe Offset does NOT work if BLTOUCH and Z_MULTI_ENDSTOPS and AUTO_BED_LEVELING_BILINEAR is enabled
(Z Probe Offset has no effect)

Z Probe Offset WORKS if BLTOUCH and Z_MULTI_ENDSTOPS and AUTO_BED_LEVELING_UBL is enabled
(Z Probe Offset works)

AUTO_BED_LEVELING_BILINEAR is BROKEN when using BLTOUCH and Z_MULTI_ENDSTOPS; 
only AUTO_BED_LEVELING_UBL works correctly with BLTOUCH and Z_MULTI_ENDSTOPS.

This bug still needs fixed.

12oclocker, 2025-08-26:
> > I finally found a workaround for the bug,
> > I have ENABLE_LEVELING_AFTER_G28 enabled, so that my bed leveling MESH will always be applied after homing.
> > //bug workaround... #define Z_AFTER_HOMING 0 //must be zero for AutoBedLeveling to work properly using a Z endstop and BLTOUCH #define EVENT_GCODE_AFTER_HOMING "G92 Z0 \n G1 Z10 F3000" //G92 Z0 is critical for AutoBedLeveling to subtract ZProbeOffset during MESH creation with BLTOUCH + Z endstop //the "G1 Z10 F3000" is simply to set the Z position after homing, since Z_AFTER_HOMING must be zero.
> > So if you are using a Z endstop and BLTOUCH, the ZProbeOffset will never be subtracted out of the MESH you create, unless you use the settings above.
> > After doing the Z probe wizard to set your Z probe offset. You can now do bed leveling and the MESH will be created and subtract out the Z probe offset correctly from the MESH. And if you change your Z Probe Offset while printing to hone in your first layer, you probably will need to rebuild your mesh again so that the MESH is rebuild with the new Z Probe Offset applied to it. This is kind of a Caveat, since if you had a single axis Z machine with BLTOUCH used for homing, adjusting the Z Probe Offset would automatically move the MESH more near to the bed, without having to rebuild the MESH.
> > Their would be a way to avoid the Caveat and have the ability to change the ZProbeOffset and have it immediatly move the MESH up or down, but for that to happen, the workaround above would need to be NOT implemented, so ZProbeOffset would not be compensated from the mesh. Then after homing and after the bed leveling is enabled, the probe.offset.z would need to be added to the Z position (but only if bed leveling is enabled) so the nozzle does not float above the bed. This method is probably the most desirable, because it would allow adjustment of the ZProbeOffset to take effect, without having to rebuild the MESH. But this option would require code changes in G28.cpp or maybe G29.cpp or maybe motion.cpp
> 
> Are you still using z endstops with this workaround?

No I am no longer using that work around. Instead use AUTO_BED_LEVELING_UBL, which is the only bed leveling that actually works correctly with Z endstops.

12oclocker, 2025-08-26:
> Why don't you remove the z endstops?

 You cannot on multi z axis AnyCubic I3 Mega. Z accuracy is never achieved because marlin software sync of z axis is not accurate enough (this is explained in detail with one of my other posts)., homing of z will take forever and accuracy consistantly fails to be achieved. Which is why the Ender dual z axis machines with bed touch sensor running marlin use a mechanical belt to synchronize the z motors.

bgolob, 2025-08-26:
> You cannot on multi z axis AnyCubic I3 Mega. Z accuracy is never achieved because marlin software sync of z axis is not accurate enough (this is explained in detail with one of my other posts)., [...]

I see. How do you configure the bed then? Do you tighten the screws all the way in and then adjust z endstops so the nozzle is 1mm above the head, and then run BL touch calibration?

12oclocker, 2025-08-26:
> > > Why don't you remove the z endstops?
> > 
> > 
> > You cannot on multi z axis AnyCubic I3 Mega. Z accuracy is never achieved because marlin software sync of z axis is not accurate enough (this is explained in detail with one of my other posts)., homing of z will take forever and accuracy consistantly fails to be achieved. Which is why the Ender dual z axis machines with bed touch sensor running marlin use a mechanical belt to synchronize the z motors.
> 
> I see. How do you configure the bed then? Do you tighten the screws all the way in and then adjust z endstops so the nozzle is 1mm above the bed, and then run BL touch calibration?

follow factory proceedure to mechanically level your bed. use mm ruler ensure bed is mechanically level from machine base on both sides, springs is about middle of compression. adjust left and right z stop sensors so they home above bed. slowly readjust until they just barely touch the bed on left and right side. do your best to mechanically level out a machine. As it is similar to a CNC machine. You need to make sure everything is as true as possible. bed leveling error values needs to be minimal and centered around the general mechanical error. if your nozzle was floating high and you performed bed leveling, all your first layers will print terrible, because it will spread that error out through 10 layers, ect.

heat bed and nozzle.
Perform normal bed tramming as if no z touch exist.
perform z offset wizard.
perform ubl bed leveling.
save settings.

12oclocker, 2025-09-03:
I coded a bugfix and submitted it to marlin, the code is here if you want to apply it yourself in the meantime... https://github.com/MarlinFirmware/Marlin/pull/28037

closed 2025-12-17 by github-actions[bot]