MarlinFirmware/Marlin #27822: [FR] Add STM32H723ZE_btt board (BTT Octopus Pro V1.1) to CI build tests

djpallax, 2025-04-28:
### Is your feature request related to a problem? Please describe.

I have been testing the latest USB changes (USBHost updates, mass storage, device mode fixes) on the bugfix-2.1.x branch with an STM32H723ZE-based board (`BTT Octopus Pro V1.1`), and encountered issues where USB was no longer functional. There is an issue for usb on #26115 
Because this board is not currently covered by Marlin's CI build tests, regressions affecting STM32H7 targets can easily go unnoticed.
Adding automated test builds for this platform would help catch potential issues early.

### Are you looking for hardware support?

Yes.
Board: `BigTreeTech Octopus Pro V1.1`
MCU: `STM32H723ZE` (STM32H7 series, Cortex-M7, 550 MHz)
Firmware Target: `STM32H723ZE_btt`

### Describe the feature you want

Please add the `STM32H723ZE_btt` target to the CI build matrix (ci-build-tests.yml). This would not only help during current USB work, but also improve future maintenance for H7-class boards in general.

Example of the line to add under STM32H7 in `ci-build-tests.yml`:
```
# STM32H7
- BTT_SKR_SE_BX
- STM32H743VI_btt
- STM32H723ZE_btt  # <-- Proposed addition
```

And the creation of the file `STM32H723ZE_btt` in the `buildroot/tests/` directory. With `SERIAL_PORT -1` i got it working. For example:

```
#!/usr/bin/env bash
#
# Build tests for STM32H723ZE_btt
#

# exit on first failure
set -e

#
# Build with the default configurations
#
# Build basic FYSETC S6 configuration
restore_configs
opt_set MOTHERBOARD BOARD_BTT_OCTOPUS_PRO_V1_1 SERIAL_PORT -1 BAUDRATE 250000
exec_test $1 $2 "BigTreeTech Octopus Pro V1.1 | Default Config" "$3"
```

### Additional context

 - PlatformIO supports STM32H723.

 - Board files and default configurations are already available in Marlin (boards.h, pins.h).

 - Adding a CI test build would avoid silent breakages like the ones I encountered when trying the latest USB stack changes.
  
 - I have tested #27815 only enabling `USB_FLASH_DRIVE_SUPPORT` and it didn't work, and also tried with `USE_OTG_USB_HOST`. Someone on the Discord channel told me that USB never worked before on this board (I have had this board only for 2 days, so I cannot relate to whether it worked previously or not).

Thanks for considering this addition!

DaLiV, 2025-04-28:
i'm also have BTT Octopus Pro, but v1.0
partially can confirm ... https://github.com/MarlinFirmware/Marlin/issues/24948 - was some usable builds but not so stable for that worked good ... and seems intersects to many other params. has not deep enough researched.

djpallax, 2025-04-28:
I want both SD card and USB flash drive to work on my printer. I realize now that I forgot to mention that I had also enabled `SDSUPPORT` during my tests.
After trying many things, I believe there might be additional places to check, for example, the `ini/stm32h7.ini` file, which may need some extra flags for proper USB compilation.
Also, I'm not sure if the Octopus pins file might need changes to fully support USB functionality.

Regardless, the important point is that current pushes to the repository are not being tested for this board and MCU.
Adding CI tests would help catch these issues earlier, which is why I opened this FR.

thisiskeithb, 2025-04-28:
> I have tested https://github.com/MarlinFirmware/Marlin/pull/27815 only enabling USB_FLASH_DRIVE_SUPPORT and it didn't work, and also tried with USE_OTG_USB_HOST. Someone on the Discord channel told [...]

https://github.com/MarlinFirmware/Marlin/issues/26115 is the feature request to add USB flash drive support for this board, so it's not a regression since it was never supported in the first place.

djpallax, 2025-04-29:
@thisiskeithb thank you for the clarification. Just to explain, my Feature Request #27822 was not directly about the USB problem itself. It was the explaination on how I discovered that the STM32H723ZE was not being included on the tests. My main goal was to request adding it to CI build tests, to help prevent future regressions across all features for H723 boards.

Even if USB mass storage never worked before, adding this board to the CI build matrix could still be valuable for early detection of compile-time errors and general firmware maintenance. In the "Describe the feature you want" section I proposed this implementation, and it should't generate any problem, because it only sets the serial port (bare minimum).

Anyway, I understand that maybe you prefer to wait until USB is functional before adding more test targets. Thanks again for your time and all the hard work on Marlin

thisiskeithb, 2025-04-29:
The SKR 3 and Biqu BX board & config are used for CI tests so that'll cover STM32H7 for now.