MarlinFirmware/Marlin #27823: [BUG] GD32 MFL does not save the configuration in eeprom.

Lazar0v, 2025-04-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

GD32 MFL does not save the configuration in eeprom.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender-3

### Electronics

GD32

### LCD/Controller

Stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[CrealityV422MFL-PRO.zip](https://github.com/user-attachments/files/19948049/CrealityV422MFL-PRO.zip)

ellensp, 2025-05-05:
no steps to reproduce...

Please provide your testing procedure

Lazar0v, 2025-05-09:
I don't see any problems with the bugfix-2.1.x version from May 5th.

closed 2025-05-09 by thinkyhead